balena-io/etcher #720: Duplicate drive objects (?) cause error in Angular view

WasabiFan, 2016-09-23:
- **Etcher version:** Beta 14
- **Operating system and architecture:** Windows 10 x64
- **Do you see any meaningful error information on DevTools?** Yes, see below.

I clicked "select drive" and this happened:

[image]

From the Dev Tools:

[image]

I only have one H drive, for the record 😆 Disk management looks like this:

[image]

WasabiFan, 2016-09-25:
I can confirm that `drivelist` is returning duplicated drives; I'm digging in further now.

jviotti, 2016-09-26:
Hey there @WasabiFan ,

Thanks for reporting this issue, its indeed very weird! Can you provide the full output of running the `drivelist` Windows script (https://github.com/resin-io-modules/drivelist/blob/master/scripts/win32.bat)?

WasabiFan, 2016-09-26:
I can't repro it right now (after waiting a few days I decided to attempt to fix it because it was preventing me from working). I do remember that the last two drives listed were both exactly the same, and `diskpart` only listed it once. `wmic` also only listed it once. Windows Explorer showed the single drive as two (as I would expect because there were two FAT partitions). I _might_ be able to re-image it into that state, but I'm not sure.

jviotti, 2016-10-24:
@WasabiFan Were the two drives reported by `drivelist` identical, or did they just shared the `device` property (e.g: `\\.\PHYSICALDRIVE3`)?

WasabiFan, 2016-10-24:
The two JSON objects were character-for-character duplicates, including size and mount point.

jviotti, 2016-10-27:
I've tried to reproduce this in every way I could think of without luck. Even with drives containing multiple partitions, it seems that Windows will only detect the first one. Do you remember which formats the partitions had? Also, did this happen after flashing a specific image?

WasabiFan, 2016-10-28:
🎉 It took some fiddling, but I can repro. The image file is this one: <https://github.com/ev3dev/ev3dev/releases/download/ev3dev-jessie-2015-12-30/ev3-ev3dev-jessie-2015-12-30.img.xz>. I verified that the hash of the file that I used to flash it is the same as a copy I just downloaded, so I assume the image is as it originally was. It looks like I slightly misremembered by testing results (or they are different now); the mount-point/name _is_ different between the two devices, however the other properties are all the same. It's also worth noting that the second drive object seems to be in some way "special" based on how in shows up in File Explorer.

[image]

https://gist.github.com/WasabiFan/4c077b78df56f0ef1795a21e35b1da21

Let me know if there's anything else I can test.

lurch, 2016-10-28:
Looks like @jviotti might need an `Exit For` after line 68 of https://github.com/resin-io-modules/drivelist/blob/master/scripts/win32.bat ?
And I guess you might also need
```
    If containsLogicalDisk = True Then
        Exit For
    End If
```
after line 69?

\*shudder\* VBScript - must be 15 years since I last looked at that.

jviotti, 2016-10-28:
@WasabiFan @lurch , 

Thank you for the amazing research on this issue, you rock! I flashed the image you pointed to, however I can only see the first partition, and can't reproduce the issue in neither of my Windows PCs.

As far as I can see, the image you pointed to contains a FAT and a Linux partition. Since Windows doesn't read Linux-specific partitions out of the box, and the fact that

> the second drive object seems to be in some way "special" based on how in shows up in File Explorer

makes me think you've installed a certain software/extension to make Windows "see" these partitions, and that such addition is needed to reproduce the issue. Any ideas?

jviotti, 2016-10-28:
> Looks like @jviotti might need an Exit For after line 68 of https://github.com/resin-io-modules/drivelist/blob/master/scripts/win32.bat ?

Yeah, that might be it, however that means only one of the mount points will be reported to the user, while in the ideal case we should append every drive letter found for the same physical drive to the same drive object.

lurch, 2016-10-28:
Would a disk / image with two FAT partitions also show up this issue?

> while in the ideal case we should append every drive letter found for the same physical drive to the same drive object.

If you do modify the script to do that, I guess the macos and linux scripts should also be updated to perform the same behaviour?

jviotti, 2016-10-28:
@lurch Both the GNU/Linux and OS X scripts collect mount points for drives containing multiple partitions. Are experiencing a different behaviour?

lurch, 2016-10-28:
I haven't actually tested them (sorry!) but from a cursory glance it looked like the Linux and OSX scripts are using just a single for-loop, whereas the Windows script is using a for-loop nested inside another for-loop, and I was assuming that that was the reason for the discrepancy that caused this issue.

jviotti, 2016-10-28:
Yeah, the implementations are smarter in those platforms:

### OS X

```sh
mountpoint=`echo "$mount_output" | perl -n -e'm{^'"${disk}"'(s[0-9]+)? on (.*) \(.*\)$} && print ",$2"'`
```

### GNU/Linux

```sh
df --output=source,target | grep "^$1" | awk '!($1="")&&gsub(/^ /,"")' | tr '\n' ','
```

And an inner loop in case we're dealing with UUIDs.

lurch, 2016-10-28:
Just out of curiosity, why the mixture of perl on OSX, and grep & awk on Linux?

jviotti, 2016-10-28:
I don't remember in detail from the top of my head anymore, but its probably because the GNU awk implementation had functionality that the OS X one lacked, they have been done by different people, or in case I did it (I don't remember honestly), I didn't check the tooling stack I was using in the other platforms. 

I'm happy to have them unified if you have enough motivation to do so!

WasabiFan, 2016-10-28:
> As far as I can see, the image you pointed to contains a FAT and a Linux partition.

Yes, this seemed odd to me too; I wouldn't expect Windows to recognize it at all. It's possible that Windows is either mis-identifying the partition or intentionally detecting that _something_ is there without understanding what it is.

> this makes me think you've installed a certain software/extension to make Windows "see" these partitions, and that such addition is needed to reproduce the issue. Any ideas?

I doubt that. You saw my opposition to 7-zip; my mentality about other crapware (or however you choose to view it) is similar. I rarely, if ever, install software that isn't from a trusted publisher, and definitely not drivers. Unless it came with something larger -- which is unlikely -- I don't think that it is a 3rd-party driver or other software that I manually installed.

One possibility is that it's something odd about the specific SD card adapter I'm using (it is a USB multi-reader) or the SD card. I can try it on a different PC to try to eliminate variables.

jviotti, 2016-10-28:
Indeed very weird, but I have good news: I'm able to reproduce by creating a drive with multiple NTFS partitions (which get all mounted by Windows). I'll fix and ping you to test if you don't mind!

WasabiFan, 2016-10-28:
Hmmm... I had assumed that it wasn't just an issue of having multiple partitions on the drive. I was under the impression that that was already something that had been used before; is this really just an issue of multiple partitions, or is there something more complicated going on?

jviotti, 2016-10-28:
@WasabiFan Looks like that is the case. I've fixed this particular problem in this PR: https://github.com/resin-io-modules/drivelist/pull/100. Can you try [the updated `win32.bat`](https://github.com/resin-io-modules/drivelist/blob/b0e9db2d791a0dcef36d3647959a99d7b1459521/scripts/win32.bat) and see if it correctly reports your drives now?

WasabiFan, 2016-10-28:
Yep, it looks like the version in your PR works as expected. The name is now Windows' device node name and the mount-point contains multiple values 🎉

jviotti, 2016-10-31:
Amazing news!

closed 2016-10-31 by jviotti