balena-io/etcher #722: Not working on Ubuntu

842Mono, 2016-09-27:
When I hit flash it asks for root password then it doesn't do anything. The button becomes "flash" again. Just that.

- **Etcher version: v1.0.0-beta.15**
- **Operating system and architecture: Ubuntu 14.04 64bit**
- **DevTools:**
    tracker.js:10 Tue Sep 27 2016 15:01:01 GMT+0200 (EET) Application start
    tracker.js:10 Tue Sep 27 2016 15:01:01 GMT+0200 (EET) Checking for updates
    tracker.js:10 Tue Sep 27 2016 15:01:54 GMT+0200 (EET) Select image ({"size":1513308160,"path":"/blablabla/ubuntu-16.04.1-desktop-amd64.iso","logo":false,"bmap":false})
    tracker.js:10 Tue Sep 27 2016 15:02:06 GMT+0200 (EET) Flash ({"image":"/blablablabla/ubuntu-16.04.1-desktop-amd64.iso","device":"/dev/sdc"})
    tracker.js:10 WRITER: Tailing /tmp/etcher-569253li5orq6Mys.log

    tracker.js:10 WRITER: Tail starting:

    tracker.js:10 WRITER: filename: /tmp/etcher-569253li5orq6Mys.log

    tracker.js:10 WRITER: options: {}

    tracker.js:10 WRITER: Attempting to elevate

    Failed to clear temp storage: An operation that depends on state cached in an interface object was made but the state had changed since it was read from disk. InvalidStateError

I found empty log files in `/tmp/`

jviotti, 2016-10-25:
I'm pretty sure this bug will be fixed with the following patch: https://github.com/resin-io/etcher/pull/774.

I'll publish a custom build here so you can test it if you can.

jviotti, 2016-10-26:
Here's the custom build: https://www.dropbox.com/s/ao1vmhhv7775mc5/Etcher-linux-x64.zip?dl=0

jviotti, 2016-10-26:
I'll close this issue as resolved, but please re-open if you can still reproduce with the custom built I sent, or with the next published Etcher version.

closed 2016-10-26 by jviotti