balena-io/etcher #723: Not Working on Debian Jessie Stable (i just use the official repository)

inukaze, 2016-09-27:
- **Etcher version: 1.0.0-beta.15**
- **Operating system and architecture: 64 Bits **
- **Do you see any meaningful error information on DevTools?**

DevTools Log : 

```
Tue Sep 27 2016 10:46:51 GMT-0400 (BOT) Application start
tracker.js:10 Tue Sep 27 2016 10:46:51 GMT-0400 (BOT) Checking for updates
tracker.js:10 Tue Sep 27 2016 10:47:08 GMT-0400 (BOT) Select image ({"size":638582784,"path":"/media/Compartido/Imagenes/Iso/Linux/trisquel-mini_7.0_amd64.iso","logo":false,"bmap":false})
tracker.js:10 Tue Sep 27 2016 10:47:12 GMT-0400 (BOT) Flash ({"image":"/media/Compartido/Imagenes/Iso/Linux/trisquel-mini_7.0_amd64.iso","device":"/dev/sdg"})
tracker.js:10 WRITER: Tailing /tmp/etcher-28846jxN9DkhvWMuT.log
Tail starting:
filename: /tmp/etcher-28846jxN9DkhvWMuT.log

tracker.js:10 WRITER: options: {}

tracker.js:10 WRITER: Attempting to elevate

tracker.js:10 Tue Sep 27 2016 10:47:32 GMT-0400 (BOT) Progress (write): 0% at 3.86 MB/s (eta 164s)
tracker.js:10 Tue Sep 27 2016 10:47:32 GMT-0400 (BOT) Progress (write): 1% at 4.89 MB/s (eta 129s)
tracker.js:10 Tue Sep 27 2016 10:47:33 GMT-0400 (BOT) Progress (write): 1% at 3.53 MB/s (eta 178s)
tracker.js:10 Tue Sep 27 2016 10:47:34 GMT-0400 (BOT) Progress (write): 2% at 4.32 MB/s (eta 145s)
tracker.js:10 Tue Sep 27 2016 10:47:34 GMT-0400 (BOT) Progress (write): 2% at 4.05 MB/s (eta 154s)
tracker.js:10 Tue Sep 27 2016 10:47:35 GMT-0400 (BOT) Progress (write): 2% at 4.45 MB/s (eta 140s)
tracker.js:10 Tue Sep 27 2016 10:47:35 GMT-0400 (BOT) Progress (write): 2% at 3.83 MB/s (eta 162s)
tracker.js:10 Tue Sep 27 2016 10:47:36 GMT-0400 (BOT) Progress (write): 3% at 4.15 MB/s (eta 149s)
tracker.js:10 Tue Sep 27 2016 10:47:36 GMT-0400 (BOT) Progress (write): 3% at 4.51 MB/s (eta 136s)
tracker.js:10 Tue Sep 27 2016 10:47:37 GMT-0400 (BOT) Progress (write): 4% at 4.27 MB/s (eta 144s)
tracker.js:10 Tue Sep 27 2016 10:47:38 GMT-0400 (BOT) Progress (write): 4% at 4.23 MB/s (eta 144s)
tracker.js:10 Tue Sep 27 2016 10:47:39 GMT-0400 (BOT) Progress (write): 5% at 4.43 MB/s (eta 137s)
tracker.js:10 Tue Sep 27 2016 10:47:40 GMT-0400 (BOT) Progress (write): 5% at 4.38 MB/s (eta 138s)
tracker.js:10 Tue Sep 27 2016 10:47:40 GMT-0400 (BOT) Progress (write): 5% at 4.58 MB/s (eta 131s)
tracker.js:10 Tue Sep 27 2016 10:47:40 GMT-0400 (BOT) Progress (write): 6% at 4.42 MB/s (eta 136s)
tracker.js:10 Tue Sep 27 2016 10:47:43 GMT-0400 (BOT) Progress (write): 7% at 4.61 MB/s (eta 128s)
tracker.js:10 Tue Sep 27 2016 10:47:44 GMT-0400 (BOT) Progress (write): 7% at 4.49 MB/s (eta 132s)
tracker.js:10 Tue Sep 27 2016 10:47:44 GMT-0400 (BOT) Progress (write): 7% at 4.51 MB/s (eta 130s)
tracker.js:10 Tue Sep 27 2016 10:47:44 GMT-0400 (BOT) Progress (write): 6% at 4.46 MB/s (eta 134s)
tracker.js:10 Tue Sep 27 2016 10:47:45 GMT-0400 (BOT) Progress (write): 7% at 4.43 MB/s (eta 133s)
tracker.js:10 Tue Sep 27 2016 10:47:45 GMT-0400 (BOT) Progress (write): 8% at 4.38 MB/s (eta 134s)
tracker.js:10 Tue Sep 27 2016 10:47:45 GMT-0400 (BOT) Progress (write): 9% at 4.7 MB/s (eta 123s)
tracker.js:10 Tue Sep 27 2016 10:47:46 GMT-0400 (BOT) Progress (write): 9% at 4.64 MB/s (eta 125s)
tracker.js:10 Tue Sep 27 2016 10:47:59 GMT-0400 (BOT) Progress (write): 9% at 4.77 MB/s (eta 120s)
tracker.js:10 Tue Sep 27 2016 10:47:59 GMT-0400 (BOT) Progress (write): 10% at 4.66 MB/s (eta 123s)
```

The eta was around 5 minutes, but i am waiting for 1 hour, and the tool don't progress anymore. and still the bar progress still is in 10% with 2:03s.

I hope this fixed soon. the interface is really simple :D and with the portability of AppImage is Awesome, i just need it, the App works fine :D 

Ok i try with the "GitHub Version" right now on the terminal first i install nodejs: 

```
sudo apt-get install -y nodejs
sudo ln -s /usr/bin/nodejs /usr/bin/node
```

Second i download the GitHub Version : 

```
git clone https://github.com/resin-io/etcher
cd /etcher/bin
./etcher
```

if this is not made when you try to use the github version, just recieve
```
$ ./etcher 
/usr/bin/env: node: There is no file or directory
```

```
$ ./etcher 
/media/Compartido/Linux/src/Utilidades/etcher/lib/cli/etcher.js:19
const _ = require('lodash');
^^^^^
SyntaxError: Use of const in strict mode.
    at Module._compile (module.js:439:25)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/media/Compartido/Linux/src/Utilidades/etcher/bin/etcher:2:1)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)

```

and well is not working for me. 
i go try the follow : 

```
$cd .. ; rm -rf build
$ mkdir -p build ; cd build
$ cmake ..
CMake Error: The source directory "/media/Compartido/Linux/src/Utilidades/etcher" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
```

Ok that don't neither.

jviotti, 2016-09-28:
@inukaze Please follow the [running locally documentation](https://github.com/resin-io/etcher/blob/master/docs/RUNNING-LOCALLY.md) for proper instructions, since you have to build the project before being able to run it.

I'm aware Debian links the `nodejs` package as `nodejs`, which is non-standard, and should be re-linked as simply `node`. Also, make sure you have at least NodeJS v6 (check `node -v`), since that version is required to compile Etcher.

Please re-open if you still hit any issues after following the documented steps!

closed 2016-09-28 by jviotti

inukaze, 2016-10-10:
Ok let me try the follow : 

```
$ node -v
v0.10.29
```

Well i uninstall the current nodejs 

```
$ ls -l /usr/bin/node
lrwxrwxrwx 1 root root 15 sep 27 11:29 /usr/bin/node -> /usr/bin/nodejs

$ sudo rm -rf /usr/bin/node

$ sudo apt-get purge -y nodejs
```

Well i need manual compile the dependencies : 

*   NodeJS (at least v6)
*   Bower
*   UPX
*   SCSS Lint

```
#NodeJS
 sudo apt-get update && sudo apt-get install git-core curl build-essential openssl libssl-dev
 git clone https://github.com/nodejs/node.git
 cd node
 git checkout v6.0.0
 ./configure --prefix=/usr
 make
 sudo make install
 cd ..

# Bower (Requieres NodeJS / NPM)
 sudo npm install -g bower
/usr/bin/bower -> /usr/lib/node_modules/bower/bin/bower
/usr/lib
└── bower@1.7.9 

#=========================================================================================

# UPX
 git clone https://github.com/upx/upx
 cd upx

# First you need UCL
# Go to -> http://www.oberhumer.com/opensource/ucl/#download
# For Download the Compression Library UCL, UPX Need it

 wget -c http://www.oberhumer.com/opensource/ucl/download/ucl-1.03.tar.gz
 tar xfvz ucl-1.03.tar.gz
 cd ucl*
 ./configure --prefix=/usr
 make
 sudo make install
 cd..
 export UPX_UCLDIR=$PWD/ucl-1.03

# Optional LZMA for Better Compresion :
# Site -> https://sourceforge.net/projects/sevenzip/files/LZMA%20SDK/4.43

 mkdir lzma443
 wget -c http://downloads.sourceforge.net/project/sevenzip/LZMA%20SDK/4.43/lzma443.tar.bz2
 tar xfvj lzma443.tar.bz2 -C ./lzma443
 export UPX_LZMADIR="$PWD/lzma443"
 export UPX_LZMA_VERSION=0x443
 make all
 cd src
 sudo cp upx.out /usr/bin/upx
 cd ../..

#=========================================================================================

# SCSS-Lint ( You need install Ruby )

#Ruby Method :
 
 git clone https://github.com/brigade/scss-lint
 cd scss-lint
 sudo gem install scss_lint
 cd ..

#== OR ==

# Method NPM :
 sudo npm install -g scsslint
```

Well i had installed all dependencies now i go try again

```
 git clone https://github.com/resin-io/etcher
 cd etcher

 uname -m
x86_64

 ./scripts/build/linux.sh install x64
 npm start
```

# Now Test with any Image : i choise "Canaima 4.1 Stable" because
i need to install it on a "Canaimita Educativa" :v 

Ready now works 100% :D

inukaze, 2016-10-10:
etcher ISO to USB Tests :

Debian 8.5.0 amd64 netinst	-> Files OK, try to boot "This is not a booteable disk. Please insert a booteable floppy and press any key to try again ..."
i open "gksu gparted" . appear the message "The driver descriptor says the physical block size is 2048 bytes, but Linux says it is 512 bytes."

Debian 8.5.0 i386 xfce CD1	-> Files OK, try to boot "This is not a booteable disk. Please insert a booteable floppy and press any key to try again ..."
i open "gksu gparted" . appear the message "The driver descriptor says the physical block size is 2048 bytes, but Linux says it is 512 bytes."

Debian Live 8.5.0 i386 lxde	-> Fail, unformated drive
Linuxmint 18 Xfce 32bit		-> Fail, unformated drive

Ubuntu-10.04.4 Desktop amd64	-> Fail, unformated drive
Ubuntu-10.10 Server amd64	-> Files OK, try to boot "Operating system not found"

Ubuntu-12.04.5 Server amd64	-> Files OK, try to boot "This is not a booteable disk. Please insert a booteable floppy and press any key to try again ..."

i open "gksu gparted" . appear the message "The driver descriptor says the physical block size is 2048 bytes, but Linux says it is 512 bytes."

ZorinOS 9 Lite 32 Bits		-> Fail, unformated drive

Some screenshots : 
https://s25.postimg.org/m1xfsf2jz/etcher_djs004_debianetinst.png
https://s25.postimg.org/jyn0kr2r3/etcher_djs004_debian_xfce_32bits.png
https://s25.postimg.org/i83zj9l7z/etcher_djs004_ubuntu10044amd64.png
https://s25.postimg.org/3q6sb9twv/etcher_djs004_ubuntusrv1204amd64.png
https://s25.postimg.org/p14cfjc1b/etcher_djs004_zorinos9b32.png

Something strange is happend here :S

jviotti, 2016-10-11:
Hi @inukaze ,

Etcher doesn't change the physical block size of your drive at all. The error you're getting comes from the image itself, usually as a consequence of hybrid ISOs (special "Frankenstein" ISOs containing certain workarounds to be able to boot in a wide range of computers, see https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos) as far as we've seen. Most Linux-specific ISO flashing tools partition the drive, mount the ISO, and manually copy files, while Etcher is just concerned with bytes.

The fact that it doesn't boot is strange though. Can you try on different computers, or flashing to different usb drives?