balena-io/etcher #725: does not build on node versions below v6

imrehg, 2016-09-28:
When building on node version below 6, modules that don't exists for the current platform cause the build to fail. For example building on Linux crashes due to `appdmg` (while I cannot seem to find, where it is used). On node 6 it successfully builds (likely ignoring uninstallable packages)

For example on node v4.6.0 compilation breaks with the following error:
```
...
make: Leaving directory '/home/greg/resin/prog/etcher/node_modules/lzma-native/build'
npm ERR! Linux 4.7.4-1-ARCH
npm ERR! argv "/home/greg/.nvm/versions/node/v4.6.0/bin/node" "/home/greg/.nvm/versions/node/v4.6.0/bin/npm" "install" "--build-from-source"
npm ERR! node v4.6.0
npm ERR! npm  v2.15.9
npm ERR! code EBADPLATFORM

npm ERR! notsup Unsupported
npm ERR! notsup Not compatible with your operating system or architecture: appdmg@0.3.10
npm ERR! notsup Valid OS:    darwin
npm ERR! notsup Valid Arch:  any
npm ERR! notsup Actual OS:   linux
npm ERR! notsup Actual Arch: x64

npm ERR! Please include the following file with any support request:
npm ERR!     /home/greg/resin/prog/etcher/npm-debug.log
```

I guess either fix build on those versions, or provide a minimum node version to build etcher.

alexandrosm, 2016-09-28:
@imrehg there is no minimum node version to build Etcher -- Each version of Etcher is targetted at a specific version of Electron, which bundles a specific version of node.js. Anything above or below that version of node.js is not assumed to work. I guess if people are interested in packaging Etcher beyond Electron, they should explore and contribute ways of having Etcher have broader support. But that is currently out of Etcher's scope.

imrehg, 2016-09-28:
@alexandrosm two issues:

* It's totally opaque and not sure where to find that info. Right now all there is a crash, and no info.
* While building I think I've seen "electron-builder" being pulled in, defining node 4.4.0, but etcher did not actually build with that version of node. So confusion ensues.

Imho this still needs either a bugfix, or clarification of some kind. Unless I've missed something, then I'd be happy to know.

jviotti, 2016-09-28:
@imrehg Originally, Etcher had to be built with the exact same NodeJS version being compiled in the specific Electron version we were using, as @alexandrosm mentioned. This enforcement has been relaxed a bit, however Etcher still requires at least v6, and there is not much we can do about it except clarifying it on the docs (will do now).

imrehg, 2016-09-29:
@jviotti that all sounds good, and really only this clarification was needed. :) Spent a couple of hours debugging yesterday why people have build errors, and just saying "this is what's supported" was the missing piece of information. Thanks very much!

closed 2016-09-29 by jviotti