balena-io/etcher #726: command line version does not elevate properly

imrehg, 2016-09-28:
- **Etcher version:** 1.0.0-beta15
- **Operating system and architecture:** ArchLinux x64

Using the command line etcher, does not seem to elevate properly after asking and getting sudo. See the log below. Explanation:

* first run, asks for sudo password, given, complains about privileges, exists
* second run, does not ask for sudo password which means the password was right the first time, should have access, complains about privileges, exists
* run with explicit sudo, does not ask for password which means indeed the password was right the first time, but now flashes card.

```
greg@nostromo:bin 18:04:22 (0) <123>
$./etcher ~/Downloads/resin-watchdog2-1.8.0-1.13.0-7ad55d1c946b.img 
? Select drive /dev/sdf (8.0 GB) - USB3.0 CRW-SD/MS
? This will erase the selected drive. Are you sure? Yes
[sudo] password for greg: 
You don't have enough privileges to run this operation.
Try running this command again prefixing it with `sudo`.

If this is not the case, and you're trying to burn an SDCard, check that the write lock is not set.
greg@nostromo:bin 18:05:15 (0) <124>
$./etcher ~/Downloads/resin-watchdog2-1.8.0-1.13.0-7ad55d1c946b.img 
? Select drive /dev/sdf (8.0 GB) - USB3.0 CRW-SD/MS
? This will erase the selected drive. Are you sure? Yes
You don't have enough privileges to run this operation.
Try running this command again prefixing it with `sudo`.

If this is not the case, and you're trying to burn an SDCard, check that the write lock is not set.
greg@nostromo:bin 18:08:01 (0) <126>
$sudo ./etcher ~/Downloads/resin-watchdog2-1.8.0-1.13.0-7ad55d1c946b.img 
? Select drive /dev/sdf (8.0 GB) - USB3.0 CRW-SD/MS
? This will erase the selected drive. Are you sure? Yes
Flashing [========================] 100% eta 0s  
Validating [========================] 100% eta 0s  
Your flash is complete!
Checksum: 88ab5b93
```

jviotti, 2016-09-28:
Its strange the CLI is asking you for the `root` password. The CLI has not elevation functionality built-in. I'll investigate. There might be a child process implicitly causing this.

lurch, 2016-11-10:
I've now dug into this and I think I've found the root(\*) cause - [umount.umountAsync](https://github.com/resin-io/etcher/blob/master/lib/cli/writer.js#L59)(\*\*) calls [utils.buildCommand](https://github.com/jviotti/node-umount/blob/master/build/umount.js#L77) which calls [elevate](https://github.com/jviotti/node-umount/blob/master/build/utils.js#L106) which then [adds](https://github.com/jviotti/node-umount/blob/master/build/utils.js#L74) a `sudo` call.

And this perfectly explains the strange behaviour @imrehg is seeing:
 * the first time he runs the etcher-cli, `sudo` hasn't cached the admin password, and so when `umountAsync` calls `sudo`, @imrehg gets asked for his password, but then the [imageWrite.write](https://github.com/resin-io/etcher/blob/master/lib/cli/writer.js#L65) doesn't have permission to write to `/dev/sdf`
 * the second time he runs the etcher-cli, `sudo` **has** cached the admin password, and so when `umountAsync` calls `sudo`, @imrehg _doesn't_ get asked for his password, but the `imageWrite.write` still doesn't have permission to write to `/dev/sdf`
 * so on the 3rd attempt, when etcher-cli is manually run with `sudo`, `sudo` has cached the admin password, and so the manual `sudo` on the command line and the `sudo` call in `umountAsync` both run without asking @imrehg for his password, and the `imageWrite.write` works as expected.

\* - accidental pun!
\*\* - which seems to be an [undocumented](https://github.com/jviotti/node-umount/blob/master/README.md) function BTW

So for consistent behaviour with the rest of etcher-cli, I suspect `umountAsync` should be modified to not automatically call `sudo`.

jviotti, 2016-11-11:
> ** - which seems to be an undocumented function BTW

The documented function is `.umount()`. When you promisify a module using Bluebird, the module creates functions having the `Async` suffix, which is the reason why you won't find `umountAsync` in the docs (is promisified in the CLI).

> So for consistent behaviour with the rest of etcher-cli, I suspect umountAsync should be modified to not automatically call sudo.

Definitely agree. I'll make a change in a minute.

> EDIT: Nowhere in the above process does it check to see if the drive is mounted before trying to unmount it, and nowhere does it check if the process is already elevated before trying to call sudo.

Oh, I was really sure we had an `isElevated()` check there. I'll fix this as well.

jviotti, 2016-11-11:
I just sent a PR to throw an error message early on if the account is not root. Happy to hear suggestions about the wording!

lurch, 2016-11-11:
> When you promisify a module using Bluebird, the module creates functions having the Async suffix

More "magic" for me to  learn ;-)

jviotti, 2016-11-18:
This issue has been fixed by https://github.com/resin-io/etcher/pull/866 and https://github.com/resin-io/etcher/issues/726. Let us know how it goes!

closed 2016-11-18 by jviotti