balena-io/etcher #729: Change the "unsafe mode" cancel button colours

lurch, 2016-09-29:
- **Etcher version: 1.0.0-beta.15**
- **Operating system and architecture: Ubuntu 14.04 x64**
- **Do you see any meaningful error information on DevTools? No**

When you click on the "Unsafe mode" in the settings screen, the following dialog pops up:
[image]

IMHO the way the "CANCEL" button is greyed-out, it looks like it's disabled and clicking on "YES, CONTINUE" is the *only* option available.

alexandrosm, 2016-09-29:
great point. And in fact, given we're trying to dissuade the user from
proceeding (to an extent) the cancel button should be

1. named differently ("keep safe mode on"?)
2. be the first option
3. be much brighter

@taahirisaacs, care to offer any design drafts?

taahirisaacs, 2016-10-25:
I've tried out 3 options.

I like @alexandrosm idea of having the button named differently.

- option 1: Darker grey button
- option 2: Change of text + dark grey button
- option 3: Change of text + brighter colored button

[image]
[image]
[image]

lurch, 2016-10-25:
Yeah, number 3 looks nice :)

WasabiFan, 2016-10-25:
I think the wording "Stay in safe mode" or even "Back to safety" sounds a bit better, but it doesn't really make a difference. However, I think the word "no" should be included in the second button. The blue is a bit awkward, but I think it communicates the point the best.

lurch, 2016-10-25:
Hmmm, and perhaps "Are you sure you want to turn this on?" could be changed to "Are you sure you want to enable unsafe mode?".

Could the "Unsafe mode" tickbox in the parent dialog remain "unticked" until the user explicitly clicks on the "Yes, continue" button?

dlech, 2016-10-25:
The "Yes" button should also be changed to say the action that it does, namely "enable unsafe mode". Otherwise, you have to think too hard about what "yes" means. As @lurch pointed out, the current question refers to "this" which is not clear at all.

jviotti, 2016-10-25:
Looks good! I propose the following based on the comments so far:

- Adopt the third design draft (red and blue buttons).
- Red button: "Enable unsafe mode"
- Blue button: "Stay in safe mode"

What do you think?

alexandrosm, 2016-10-25:
I share @wasabifan's concerns, and really like the amount of attention
we're collectively putting to this.

- Keeping the checkbox off until the user has confirmed would be a nice
touch.
- "Enable Unsafe Mode" and "Stay in Safe Mode" is probably a good,
symmetric pair. Briefly considered "Enter Safe Mode" to make it even more
symmetric, but it harms the meaning, I think.
- WRT colours, I think grey is fine and would like to avoid adding a new
one to the pallet.

alexandrosm, 2016-10-25:
is the blue button part of our palette?

also, what about not checking the box before the user confirms?

jviotti, 2016-10-26:
> is the blue button part of our palette?

Yeah it is, although I agree with you that gray looks better. Blue and red are both very prominent colors and make it hard to identify which is the "primary action".

> also, what about not checking the box before the user confirms?

Sounds good!

lurch, 2016-11-03:
I've been having a play about with etcher, and when you have more than two USB drives connected, you get a "Select drive" button. When you click on it a dialog pops up and the "CONTINUE" button initially has a light-gray background, you can't click on it, and it looks just like the "CANCEL" button in the screenshot in my initial comment on this thread. When you actually **do** select a drive, the button changes to a blue background, and it can then be clicked on (which as expected closes the dialog).

So despite people's concerns about a blue button being "too intense", there *is* already precedence in Etcher for a grey-button meaning disabled, and a blue-button meaning clickable.

WasabiFan, 2016-11-03:
I think the difference there is that in the "select drive" case, there's only a single button (one option), while in the warning dialog there is another button with a bright color that has a specific meaning. Two options with bright colors makes it hard to discern a "primary" button to focus on.

lurch, 2016-11-03:
For this specific dialog, I'm not sure there **is** a "primary" button / action?
 * If you're a novice user, and you accidentally click on the unsafe tickbox, then your "primary action" is to remain in safe mode
 * If you're an advanced user, and you really do want to enable 'unsafe mode', then your "primary action" is to enable unsafe mode.

In either case, the blue button is the 'safe' option, and the red button is the 'dangerous' option.

WasabiFan, 2016-11-03:
Sorry, that was a bad choice of words. I was referring to "primary" in the visual sense, where one button draws the eye naturally. Having two of them makes it visually confusing when the dialog appears. I agree that the "primary" logical option depends on the user.

jviotti, 2016-11-03:
The dialog containing both a red and blue button looks very weird to me, but happy to go with it if everyone agrees is the best choice. Alternatively, we can put an "X" in the top right corner of the dialog and keep a single button, although it might not be very straightforward how to "cancel".

lurch, 2016-11-03:
How would it look if "ENABLE UNSAFE MODE" was displayed in dark grey (NOT the light grey that "STAY IN SAFE MODE" uses currently), and "STAY IN SAFE MODE" was displayed in blue?
@taahirisaacs @alexandrosm 

I guess it's safer to imply that "STAY IN SAFE MODE" is the primary button? (even if it means we can't display the "ENABLE UNSAFE MODE" in 'dangerous red')

closed 2016-11-07 by jviotti