balena-io/etcher #731: Display "nicer" error message when selecting an invalid zipfile

lurch, 2016-09-29:
- **Etcher version: 1.0.0-beta.15**
- **Operating system and architecture: Ubuntu 14.04 x64**
- **Do you see any meaningful error information on DevTools?**
[image]

If I select a zip file, which **doesn't** contain a disk-image file, then I get this rather ugly error message:
[image]

It would be nice if you could catch this error, and display a nicer-looking error message :-)

jviotti, 2016-09-30:
Ugh, it's indeed very ugly. The stack trace shouldn't be coming through in this case. I wonder if there was a change on the error processing and showing logic that changed recently and caused this regression. 

I'll investigate and report back!

jviotti, 2016-10-07:
Fixed in https://github.com/resin-io/etcher/pull/752

closed 2016-10-07 by jviotti