balena-io/etcher #734: "rawr i'm a dinosaur" when writing bz2 file

imrehg, 2016-09-30:
- **Etcher version:** git head at e47a70606ce017a14a3ab29a1edeee5e6399b3b1
- **Operating system and architecture:** ArchLinux x64
- **Do you see any meaningful error information on DevTools?**

```
tracker.js:10 Error: rawr i'm a dinosaur
    at ChildProcess.child.on (/home/greg/resin/prog/etcher/lib/src/child-writer/index.js:92)
    at emitTwo (events.js:106)
    at ChildProcess.emit (events.js:191)
    at handleMessage (internal/child_process.js:718)
    at Pipe.channel.onread (internal/child_process.js:444)
a.(anonymous function)                       @ tracker.js:10
(anonymous function)                         @ /home/greg/resin/prog/etcher/node_modules/angular/angular.js:13708
logException                                 @ /home/greg/resin/prog/etcher/lib/gui/modules/analytics.js:157
reportException                              @ /home/greg/resin/prog/etcher/lib/gui/modules/error.js:45
ImageWriterService.flash.then.catch          @ /home/greg/resin/prog/etcher/lib/gui/pages/main/controllers/flash.js:82
processQueue                                 @ /home/greg/resin/prog/etcher/node_modules/angular/angular.js:16170
(anonymous function)                         @ /home/greg/resin/prog/etcher/node_modules/angular/angular.js:16186
$eval                                        @ /home/greg/resin/prog/etcher/node_modules/angular/angular.js:17444
$digest                                      @ /home/greg/resin/prog/etcher/node_modules/angular/angular.js:17257
(anonymous function)                         @ /home/greg/resin/prog/etcher/node_modules/angular/angular.js:17483
completeOutstandingRequest                   @ /home/greg/resin/prog/etcher/node_modules/angular/angular.js:5955
(anonymous function)                         @ /home/greg/resin/prog/etcher/node_modules/angular/angular.js:6234
d.util.hasFunction.f.(anonymous function)    @ tracker.js:3
```

Get this `rawr i'm a dinosaur`error while writing the `bz2` compressed version of the Raspbian Jessie image. Very slow writing (~3x slower than other compressed images), and breaks around ~95%.

[image]

Happens both with `1.0.0-beta15` AppImage, and git head `npm start`.

jviotti, 2016-09-30:
Haha wtf? This must be coming from the bz2 decompression module we use (can't think of anything else). Could be a funny way to say a package is deprecated?

jviotti, 2016-09-30:
Indeed: https://github.com/regular/unbzip2-stream/blob/845023bb9b032706b6527704334cb0403a865f6c/lib/bzip2.js#L274

Do you mind creating a new issue on that project, probably linking to the problematic image? The maintainer is very responsive and I honestly don't have the knowledge to debug such as bz2 algorithm issue.

imrehg, 2016-09-30:
Sure thing :) Filed bug: https://github.com/regular/unbzip2-stream/issues/10

regular, 2016-10-19:
PR would be appreciated though!

jviotti, 2016-10-20:
No worries! I'll study how bzip2 works, and will try to fix or at least make sense out of it whenever I can allocate some time for it!

WasabiFan, 2016-10-20:
@jviotti @lurch actually seems to have figured it out already: https://github.com/regular/unbzip2-stream/issues/10

alexandrosm, 2016-12-02:
@jviotti where are we with this one?

jviotti, 2016-12-02:
We have https://github.com/cscott/compressjs, which is supposed to have
a fix for this issue, however they don't expose a standard NodeJS
implementation (they roll their own, which I can't get to play with
NodeJS, since its all done inside a single function), so we don't have a
good bzip2 decompression alternative yet.

lurch, 2016-12-02:
Maybe it's worth nudging the author of `unbzip2-stream` and asking him to pick up the bzip2 fixes from `compressjs` ? I could do that if you like?

jviotti, 2016-12-05:
@lurch That would be amazing! Thanks a lot!

regular, 2016-12-05:
@lurch Even better than nudging: a PR with test. I don't use that module in any of my projects and I am no expert on bzip at all, so I'm in no way more qualified to apply that patch. I will however merge a PR asap and republish the npm module.

lurch, 2016-12-05:
> Even better than nudging...

Haha, I'll see when I have time to put something together :smile_cat:

alexandrosm, 2017-03-02:
@jviotti how much work would be to either port the fix or use another library? This is our second oldest bug :)

jviotti, 2017-03-02:
I started hacking on it at some point, got too deep into the rabbit hole, and ended up abandoning it. I'm happy to give it another go.

alexandrosm, 2017-03-02:
I guess we could "won't fix" this. Not sure it's worth your time.

jviotti, 2017-03-03:
I guess there's no reason to not keep it open until a clearer way forward appears.

regular, 2017-03-21:
This might be fixed in  bunzip2-stream@1.011 Let me know if it is!

jviotti, 2017-03-21:
Amazing stuff @regular , thanks a lot for this. I can't find any image that reproduces the issue, so I'll upgrade this on Etcher, and we'll see if it ever happens again, unless @imrehg still has the problematic image somewhere.

lurch, 2017-03-21:
Unfortunately I no longer have a copy of @imrehg 's `2016-05-27-raspbian-jessie.img.bz2` file - I must have deleted it due to lack of disk space :-(  (and @imrehg 's dropbox download link is no longer working)
So I instead attempted to recreate it by downloading, extracting and recompressing http://downloads.raspberrypi.org/raspbian/images/raspbian-2016-05-31/2016-05-27-raspbian-jessie.zip and that **did** then recreate the infamous dinosaur-error with `unbzip-stream@1.0.10` ;-)
And upgrading `unbzip-stream@1.0.11` _did_ then eliminate the bug, so I can confirm v1.0.11 fixes this problem :smile:

closed 2017-03-21 by lurch

jviotti, 2017-03-21:
Amazing news!!