balena-io/etcher #738: Error: Command failed

Olivier3869, 2016-10-01:
- **Etcher version:** 1.0.0-beta.15
- **Operating system and architecture:** Linux Mint 18 
- **Do you see any meaningful error information on DevTools?**

{ Error: Command failed: /usr/bin/sudo -n -E -- sh -c 'mkdir -p /tmp/.mount_0MCj2p-elevated && mount -o loop -o ro /home/olivier/Documents/Etcher-linux-x64.AppImage /tmp/.mount_0MCj2p-elevated && env ELECTRON_RUN_AS_NODE=1 ETCHER_INTERNAL_LOG_FILE=/tmp/etcher-20612vdxJVz3egp9o.log /tmp/.mount_0MCj2p-elevated/usr/bin/etcher /tmp/.mount_0MCj2p-elevated/usr/bin/resources/app.asar/lib/src/child-writer/writer-proxy.js /tmp/.mount_0MCj2p-elevated/usr/bin/resources/app.asar /media/olivier/Softs/Images\ ISO\ d\'OS/Antivirus/kav_rescue_10.iso --robot --drive /dev/sde --unmount --check --log /tmp/etcher-20612vdxJVz3egp9o.log ; sleep 1 ; umount /tmp/.mount_0MCj2p-elevated'
/bin/sh: 1: Syntax error: Unterminated quoted string

    at ChildProcess.exithandler (child_process.js:214:12)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:850:16)
    at Socket.<anonymous> (internal/child_process.js:323:11)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:492:12)
  cause:

lurch, 2016-10-01:
Looks like this is caused by the fact you have a `'` in your filepath - I guess https://github.com/resin-io/etcher/blob/master/lib/src/child-writer/writer-proxy.js#L143 needs to fixed to properly escape quote characters.

jviotti, 2016-10-03:
Good catch @lurch !

ilansavir, 2016-10-18:
Is there a solution for this?

it happens randomly

jviotti, 2016-10-18:
@ilansavir, I haven't had time to look into this yet, but if @lurch 's theory is correct, removing the quote or any other special character from the image name should be enough.

lurch, 2016-10-18:
...where "the image name" also includes the full-path to the image, including any parent folders ;-)

jviotti, 2016-10-19:
Haha, great catch again! :D

jviotti, 2016-10-26:
This issue has been fixed by https://github.com/resin-io/etcher/pull/775

closed 2016-10-26 by jviotti