balena-io/etcher #739: etcher.io issues on Firefox

tsjnachos117, 2016-10-01:
When visiting https://www.etcher.io/ on Firefox, the drop-down menu next to the "Download Now" button didn't work, preventing me from downloading anything but the Windows version of the app. Instead of getting versions for different operating systems, I got an empty menu. Please fix this, as I generally don't trust Chrome/Chromium (Google loves to spy on everyone).

Also, there's a big empty space between the top of the webpage (where the "Download Now" button  the "ETCHER BY RESIN.IO" are located), and the gif showing what the app looks like. This empty space has actual stuff in Chromium, which means Firefox isn't rendering something correctly there, either.

[Here](https://mega.nz/#!VgI23BZK!1F8jckPa9XBWZ4RCFTAB4O5eOaEuvHh3VBzpNLwlHkM) is a picture of what the top of the webpage looks like with Firefox (I had to upload it on mega.nz, since GitHub gave me errors everytime I tried to attach it to this issue report). [Here](https://mega.nz/#!woA2FQSZ!R2iP3u8k75KwnJVc_09LgeUVm3jWZPpZM7Lxvd-45wg) is another picture of what it should look like (or, at least what it currently looks like in Chromium). I didn't take a shot of the download menu in Chromium (mostly because I didn't think about it until after I uploaded the picture, and by then I didn't care to upload another), but it works just fine.

PS: I wasn't even using Windows when visiting this site, which must mean the site wasn't reading my Firefox user agent correctly. Everything worked fine in Chromium (which, as I mentioned, I don't like using).

tsjnachos117, 2016-10-01:
Also, the same thing just happened again when I visited with QupZilla, a webkit-based browser. I would assume other webkit-based browsers, such as Apple Safari would also have similar issues.

alexandrosm, 2016-10-01:
paging @craig-mulligan

craig-mulligan, 2016-10-02:
Hey @tsjnachos117, thanks for reporting. I tried on firefox, QupZilla and Safari but I'm was unable to reproduce. Do you mind checking you dev tools console for errors?

Also could you give it a try with all your firefox plugins disabled?

davidak, 2016-10-18:
I also can't reproduce this in Firefox 49 on OS X.

@tsjnachos117 maybe you have some Extension that blocks all JavaScript?

jviotti, 2016-10-31:
Thanks a lot for your help @davidak @craig-mulligan! Looks like a plugin/extension issue indeed. I'll close it unless the reporter can provide any more information.

closed 2016-10-31 by jviotti

lurch, 2016-11-02:
Yeah, I was able to reproduce this exact behaviour in Firefox using the [RequestPolicy](https://addons.mozilla.org/en-US/firefox/addon/requestpolicy/) plugin, if I didn't allow access to the amazonaws.com server. Problem easily fixed by allowing access from the RequestPolicy dropdown menu :)

tsjnachos117, 2016-11-10:
I think I've figured out the issue. I do have a content blocker extension, which was blocking some resources on **https://resin-production-downloads.s3.amazonaws.com**.  Telling uBlock Origin not to block that domain appears to have fixed the issue.

If anyone else visits that domain with ad blockers enabled, try unblocking the above domain. Maybe that will help.