balena-io/etcher #740: Allow publishers to embed metadata, enhancing etcher functionality and UX

alexandrosm, 2016-10-02:
Etcher's mission has been to empower users to flash images with a minimal amount of effort while using intelligent extraction of context from the environment to prevent common failure modes.

There are a few more things we can do to improve on this mission, but those too will soon run out if we continue at the current pace. The next frontier for improving functionality and user experience is to create a channel whereby image publishers can convey additional information about the image being written. Etcher can then utilise that metadata to optimise its current functionality, and eventually provide additional features to users.

# Types of extended metadata

Some examples of what kinds of metadata could be bundled:

- Logos, links, descriptions about the publisher, intended device(s), and the image itself.

On the most basic level, additional descriptive metadata can help Etcher customise its appearance, inform the user, and generally provide an experience that fits better with the workflow Etcher is being used for, both stylistically and functionally.

- Image bmap

[bmaps](https://source.tizen.org/documentation/reference/bmaptool/introduction) are an exciting technology allowing image writers to "skip over" meaningless areas of the image (e.g. what would normally represent empty space for the filesystem being written). Indicatively, a bmap can save anywhere from 15% to 60% of the time used to write, though due to the nature of the technology, that number could theoretically range anywhere from near-0% to near-100%.

- Configuration metadata

Assuming there was a reliable way to map a fairly simple set of key-value pairs into various configuration files in a filesystem, the extended format could include those mappings, allowing Etcher to present a fairly simple ui to the end-user, and then use that data to configure the image. Image configuration is particularly important for images aimed for headless devices that give their users very little opportunity to interact with them once booted.

- Image update checking

A common problem is that an image that is downloaded may be used much later, when a newer version exists. That newer version may have fixed an important bug or simply be better aligned with the documentation. Having a way to notify a user that there is a newer version of the image selected would help avoid many negative experiences.

There are, of course, many more things that can be done with extended metadata, but each of the four types described here is both compelling and feasible in the short term, some of them having been already partially or fully implemented in Etcher.

# How do we embed?

Making a new format is always a big decision, and the kind of ecosystem fracturing it entails is painful, often preventing adoption. In the case of OS images however, there is a loophole we can use to convey this metadata without forcing publishers to use between etcher and all other writers.

Since images are often delivered as archives, we can simply require an additional folder, with a name that users would be inclined to ignore, exist in the archive. That folder could contain all sorts of assets and metadata, leaving the root of the archive to the image. A user can still perform their normal workflow of uncompressing and writing the image with any other procedure, but if the archive is opened with Etcher, then all the extended functionality can be activated.

We could still provide an extension for this format in case publishers explicitly *want* to prevent use of other writers, e.g. `.etch`, but the method itself should not require this extension be used.

dlech, 2016-10-02:
As a vendor of images for multiple embedded systems, I particularly find the "intended device" feature potentially useful. But I would like to point out that many of our users don't even know what "Raspberry Pi" or "BeagleBone" are, so having a picture of the intended device would be even more useful than just the name. It has been my experience so far that people will try to, for example, install a BeagleBone image on another device even if I write "THIS IMAGE IS FOR BEAGLEBONE ONLY" in all capital letters on the download page.

alexandrosm, 2016-10-02:
Any thoughts as to how to convey the intended device as clearly as possible in the UI? As you say, users may not be able to cleanly distinguish a beagle from an rpi, nevermind a rpi3 from an rpi2 (which also look almost identical visually).

dlech, 2016-10-02:
One possibility is to use my suggestion from #742. The info popover or dialog window could show this image.

Another possibility is in the proposed configuration tool, you could allow images as part of the configuration options. For example, it might have an option where you choose which add-on board you are using and include a picture to help decide which one.

To distinguish between hardware revisions, there might be close-up pictures of where the hardware version is printed on the circuit board.

Furthermore, extended metadata may even contain pictures of similar but unsupported devices. This way, in the configuration tool, if the user selects the unsupported device, then we can display a big red X and a message explaining why the device is unsupported or where to download the correct file for this device.

---

The real-life example of what we have:

We have 2 images for Raspberry Pi. One based on Raspbian and one on official Debian armhf. The official Debian image only works on RPi2/3.

These images support 3 addon boards. BrickPi, BrickPi+ and PiStorms. The BrickPi uses the UART that is also used by Bluetooth on RPi3, so we need to take this into consideration.

So the configuration tool flow in etcher might look something like this (using the official Debian image as an example):

* Select the RPi model. This could be radio buttons with a picture of each RPi model number printed on the circuit board.
* If the user selects any model 0 or 1 version, show an error message with the download link for the Raspbian image. Prevent any further configuration.
* Select the addon board. Since BrickPi and BrickPi+ are visually similar, show pictures of the hardware revision printed on the circuit board. 1.7.3 == BrickPi, 2.8 == BrickPi+. There is actually PiStorms and PiStorms-v2 as well. There is no configuration difference, but we might want to show pictures of both anyway.
* If the user selected RPi3 and any BrickPi, then show an option for Bluetooth to be disabled or to use the miniuart with reduced performance.

We have a similar situation with BeagleBone:

* Select the BeagleBone model (White, Black, Green, WiFi, etc.)
* If the user selects BeagleBone Green Wireless, show a message that explains that this board will not work because of pin conflicts and a different model of BeagleBone will be needed.
* Select the addon board, FatcatLab EVB or QuestCape.

alexandrosm, 2016-10-02:
Thank you for the example. I really like the idea of selecting the actual board type in the configuration step. In fact, since we'll be making "datatypes" for some common/complex cases, this could be one of them, where the user can follow a wizard to determine what type of device they're looking at, similar to the guide ubuntu uses to try and figure out what your keyboard layout is by asking you to press various buttons.

dlech, 2016-10-02:
I suppose you could even take this a step farther and have a metadata-only file. This would allow us to have a single download that works for all of our devices.  As part of the configuration wizard, as you suggested, it could show all of our supported devices and then decide which of our 4 actual disk images to actually download and burn.

alexandrosm, 2016-10-02:
We had thought about a metadata-only file that has a url or torrent reference to the image, but not the idea of using the configuration step to decide which image to use. Such a great idea! Need to think about how that would be integrated into the flow without being too disruptive, but it's obvious that it's the right direction to go in, especially considering some other skinning/installer ideas we've been floating around (need to write those up too)

lurch, 2016-10-02:
Yeah, interesting ideas :-)
Perhaps images of common components (various Raspberry Pi models, Beagleboards, etc.) could be included in Etcher by default, and images of less common components (BrickPi, BrickPi+, QuestCape) could be included in the extended archive. But I guess this is something that needs to be thought about carefully, to avoid feature creep ;-)
(And the metadata in the extended archive should specify whether this "board selection mode" should be used or not, to prevent it "getting in the way" for publishers that don't want to use it)

For people who want to use Etcher 'offline' (or who want to repeatedly flash the same image to multiple SD cards), in addition to having a "metadata only" extended archive (which points to 4 different URLs), maybe you could also have an extended archive which contains all 4 of the disk images. As you suggested in #707, perhaps a custom file-extension should be used if the extended archive contains 0 or >1 disk images, with the `.zip` extension reserved for cases where 1 disk-image is included?

alexandrosm, 2016-10-02:
The format, as envisioned in my conversations with @jviotti (but not necessarily as written down, working on it) should have no problem working with one or more images, local or remote. It should be trivial to get Etcher to infer the right strategy to use. I suppose using either .zip or .etch should be also fine, and the publishers can decide which way to go, though if there are no images included, the .zip is less useful.

For further "offline" functionality, the current form of resin-cli (which is the non-generalised form of etcher-cli plus more things) currently can download images (from resin.io) and uses a local cache to avoid re-downloading all the time. May be interesting to consider, but I do worry about accusations of bloat.

As for which images to bundle or not, I'm increasingly thinking we should build a central registry of metadata that Etcher and the archives will draw from. Preloading only certain "popular" images means that we'll still have gaps in offline mode, but only some and not others, which still feels messy and complex. And I'd hate nothing more than us having to be the referees about what is popular and what isn't. I'd rather make it clear that archives that want certain assets available offline can bundle them and leave it at that until we figure out a clear story. I guess if the size of the images is huge we should rethink that, but then again people are already complaining about Etcher's size. (it's like the #1 complaint from the optimizationheads).

Thinking about the configuration/image selection problem, one fairly simple way to get that done would be to make it possible to define configuration pre-requisites for specific images to be valid, such that Etcher can infer that it needs to extract the configuration value before deciding which image to use. In the regular case where no pre-requisites are defined, the normal workflow proceeds. 

This is actually touching on some things I was pondering during my PhD on constraint-based modelling.. I did *not* expect Etcher to get there this soon. Fascinating, lots to think about.

WasabiFan, 2016-10-02:
> Preloading only certain "popular" images means that we'll still have gaps in offline mode, but only some and not others, which still feels messy and complex.

Make it simple: take the thumbnails that you create for Resin.io-supported devices and make them available for download by others. That way, you aren't forcing vendors to create their own pictures, but at the same time maintenance is minimal. When a vendor wants to create a RasPi image, for example, they can just download and include the pre-made thumbnail that you provided. Maybe you could even allow people to open PRs against a repository of images to add ones that they create.

lurch, 2016-10-02:
Regarding (photo) images being built-in to Etcher: I was thinking it'd be more consistent if each extended archive that wanted to e.g. do the Pi-photo-based model-selection / configuration each used the same Pi-pictures, rather than each publisher including their own (possibly different) images? And WRT Etcher deciding which boards are "popular", the whole thing should be extensible so publishers can include their own board-images, if their particular board-image isn't included in Etcher yet. (and hopefully they'd then file a PR against Etcher, requesting it be added to the next release)

Or was your "central registry" idea a suggestion that resin.io could offer a repository of suitable images, which publishers would be free to include in their extended archives?

> For further "offline" functionality, the current form of resin-cli (which is the non-generalised form of etcher-cli plus more things) currently can download images (from resin.io) and uses a local cache to avoid re-downloading all the time.

Is the cache optional, if people are running low on disk-space? (although obviously the cache would be mandatory if you wanted to do a torrent-download)

Perhaps I'm just "oldskool" in preferring big offline installers (which I store on my NAS), rather than assuming that my internet connection will be reliable enough to do the downloading on demand with an "online" installer. But if course this is all up to the publishers, as long as Etcher supports it.

> This is actually touching on some things I was pondering during my PhD on constraint-based modelling.. I did _not_ expect Etcher to get there this soon.

LOL :+1:

alexandrosm, 2016-10-03:
Essentially my suggestion on the repository is that we create a repo
somewhere around here but separate from Etcher itself that acts as a
structured wiki for device type data. The device repos on the resin-os org
actually have a healthy amount of metadata, and refer to even more metadata
provided by the yocto project, but we've learned with Etcher that it is a
lot better if we take what we've learned from resin.io specific solutions
and rephrase it so that it's useful to the wider community.

The one thing that scares me is how we'll deal with a problem that we've
faced pretty hard on resin.io so far, which is defining what is a "device
type" and what is a "device type variant". Much like defining what is and
isn't a "country", most people think it's obvious, but when they get into
the weeds, they realise that there's more nuance than they initially
thought. All this said, I still think it can be done if we inject just a
little bit of "editorial opinion", we just have to be mindful of the
pitfalls.

alexandrosm, 2016-10-03:
As to whether we bundle the repo with Etcher, I'm just a bit worried that the size of the metadata will grow (especially as we start adding photos) and that will dynamite our efforts to keep Etcher's size under control. If all that is available for download and embedding into each archive as @WasabiFan mentions, we should all be ok, and the offline mode should be intact.

For the offline mode, perhaps we can learn from Google Maps' approach to offline usability. You can explicitly ask it to make you a part of the world available offline. Similarly, one could ask Etcher to "inflate" a certain archive by downloading all the files it references into it.

Though in principle I have no objection to such functionality, I do wonder if it will ever rise to top priority status. On the other hand, if we leave room for it and people do really want it enough to build it, I don't see why we'd object to a PR like that.

lurch, 2016-10-03:
Yeah, there's similar "weeds" when trying to define Languages or Timezones ;-)

Would the "SKU" concept be useful? And I guess this is also something where the publishers should be allowed to add their own levels of "device differentiation"?

alexandrosm, 2016-10-03:
One of the best device support experiences we had was when we worked with the Parallella team, and they were crystal clear on what editions of their board had been made. A lot of the confusion comes from trying to piece together information that's out there in blogs etc. If we manage to attract device manufacturers, half the problem is solved. 

Care to say more about the SKU concept?

lurch, 2016-10-03:
The "inflate" idea is interesting, but I'd be a bit wary about Etcher modifying the original extended archive file. I guess it might lead to misleading filenames, if the file was called something like `ev3dev_no_images.etch` ? I've no problem with a "download all" button which put all the images into e.g. a subfolder though.

alexandrosm, 2016-10-03:
I see what you mean. I suppose it could create a new file with the result of the inflation, adding something like -inflated to the filename. Assuming the deflated .etch should be a fairly small file, the duplication should be minimal, and the clarity of the result worth it.

lurch, 2016-10-03:
> Care to say more about the SKU concept?

https://en.wikipedia.org/wiki/Stock_keeping_unit
When I was in the game-developer world, we'd have an American SKU, a European SKU, a Far East SKU, etc. (i.e different versions of the game-disc)

jviotti, 2017-01-24:
I'm closing this given the new approach we're going to take towards image metadata.

closed 2017-01-24 by jviotti

WasabiFan, 2017-01-24:
I've seen this mentioned as the close reason for multiple issues... What is this new approach?

jviotti, 2017-01-25:
Hi @WasabiFan ,

Sorry for not being explicit enough (I think I described things better in a couple of issues but then become lazy about it). The idea is that the image metadata will live in a centralized metadata repository. By decoupling the metadata from the image itself, we solve a lot of streaming related issues, simplify the whole model, and allow Etcher to become some sort of "marketplace", where we will provide a way to browser our "catalog" of officially supported images (with metadata) and download & flash directly from it.

This is all being heavily discussed at the moment, but I'll be creating a stream of new GitHub issues as a new GitHub milestone to kickstart development very, very soon.