balena-io/etcher #742: Use info icon button instead of "SHOW FULL FILENAME"

dlech, 2016-10-02:
[image]

Info icon buttons are becoming a fairly ubiquitous way to indicate that detailed information is available if you click the icon.

For images that contain extended metadata (#740), this could be a way to display some of the informational items. For images without extended metadata, clicking this icon could show the full file name and the size of the file.

alexandrosm, 2016-10-02:
sounds good to me. The long text there always bothered me. @jviotti @taahirisaacs, thoughts?

dlech, 2016-10-02:
I almost forgot... Another important part of my suggestion here is that the text in the info window can be copied to the clipboard. This way, when we ask someone which image they are using, they can just copy the info from here and paste it somewhere.

jviotti, 2016-10-03:
Sounds good! Regarding copying text to the clipboard, the reason why it doesn't work at the moment is that text selection collides with the fact that we allow the window to be dragged from anywhere, but maybe there is a way we can allow this in certain element (e.g: a CSS class that when put on an element, disables dragging and enables text selection).

Maybe we should create a separate ticket for that?

jviotti, 2016-10-03:
Regarding showing extended metadata information, we could display a list saying what is available, similarly to what we currently output on DevTools when an extended archive is selected. For example:

- Bmap (tick logo)
- Logo (tick logo)
- Etc.

lurch, 2016-10-03:
Or you could have a "copy to clipboard" button?

jviotti, 2016-10-03:
That mitigates the issue but doesn't solve it completely. The fact that people can't select text has been a recurrent complaint that applies to several sections of the application, and unless we can to put dozens of "copy to clipboard" buttons all over the place, we must find something that scales.

A certain CSS class that gets applies to certain areas (like in modals, by default, for example), should be good enough I think.

dlech, 2016-10-03:
> Regarding copying text to the clipboard, the reason why it doesn't work at the moment is that text selection collides with the fact that we allow the window to be dragged from anywhere, but maybe [...]

A copy to clipboard icon that copies everything might be OK too.

lurch, 2016-10-03:
Isn't showing the presence / absence of bmap files etc. more of a 'developer feature' (for publishers) though, rather than an end-user feature? (who shouldn't need to know or care what bmaps are)

alexandrosm, 2016-10-03:
We'd discussed having a "turbo mode" in the write progress bar, maybe a
lightning icon, that provides some hook for interested users to know what's
going on, but other than that, then yeah, we shouldn't be exposing that
much to the user.

jviotti, 2016-10-03:
Yeah, good points, lets only display the full image name (and size potentially).

closed 2016-10-24 by jviotti