balena-io/etcher #743: Etcher should help discover the devices once they've booted

alexandrosm, 2016-10-02:
Device discovery is another messy problem users have after burning an sd card successfully. Many guides recommend using nmap, but that is way too much of a deep dive for many users. Assuming the device boots up in the same area as the machine Etcher is running on, Etcher should have ways to help discover it. 

There could be two approaches to discovering devices. 

* Passive: some on-device code runs on the OS that exposes a pattern that Etcher is configured to detect. (e.g. mdns or other). In fact, there are ways in which we can discover a cooperating device in the local area, even if that device is not connected to a network at all. (Black magic that the resin.io team had implemented way back in the day and we could dig out if needed).

Pros: Etcher doesn't have to bear the complexity of network scanning or other approaches. Additional functionality may be provided by the on-device agent, like a web server that users can use to interact with the device.

Cons: Requires the OS to be modified in a specific way for Etcher to discover it.

* Active: Etcher scans the network and discovers new devices as they appear. Perhaps the extended archive could contain some information about e.g. the hostname that the device will use. If such code is implemented, must make sure no network scan information is transmitted beyond the users' device in any case.

Pros: does not require device OS modification, can work with nearly any distro.
Cons: requires Etcher to encode all sorts of heuristics to guess what may be a device or not, as all sorts of things may pop into or out of a network at any time. May be considered intrusive if the network scanning is extensive?

I suspect we may have to implement parts of both approaches, with the "Passively" discovered device having the advantage of additional functionality becoming available as the agent progresses.

dlech, 2016-10-02:
+1 for mdns/zeroconf

Although then it is really not Etcher's job to detect the device. It is up to the ssh program to find devices advertising ssh services and the web browser to detect devices advertising web servers, etc.

dlech, 2016-10-02:
Also, in the case of ev3dev, quite often there is no network connection after boot. Since the devices are robots, there is generally no wired ethernet. Wi-Fi may work at boot if there was a way to configure it with Etcher. For USB and Bluetooth though, we generally have to setup Internet connection sharing on a host computer before we get a network connection.

alexandrosm, 2016-10-02:
As a random question, if Etcher was able to help with the connection sharing, would that be useful? (assuming we have all the right permisisons from the user, etc for the moment)

dlech, 2016-10-02:
Yes, that could be useful.

lurch, 2016-10-02:
Off the top of my head, there's two ways that a Raspberry Pi running Raspbian could be detected:
 1. It advertises the mdns hostname `raspberrypi.local`
 1. It has a MAC prefix of `B8:27:EB`

The former is obviously a bit problematic if you have multiple Raspberry Pis on your network, and you haven't bothered giving them different hostnames yet ;-)

I dunno how cross-platform it is, but you could also try looking at the ARP cache?

A cross-platform way to setup connection sharing would indeed be very useful - have a look at the comments on http://blog.gbaman.info/?p=791 to see some of the problems people have.