balena-io/etcher #744: [feature] Confirm exiting application while flashing image

waspinator, 2016-10-02:
- v1.0.0-beta.15 - 2016-09-26
- macOS x86

While an image is being flashed, it's too easy to close Etcher unintentionally. I believe it would be a helpful feature to show a `confirm()` dialog before quitting while an image is being flashed.

jviotti, 2016-10-03:
Great idea! I'm sure Electron provides a way to intercept a close event. The `FlashStateModel` component also provides a nice `isFlashing()` method that can be used to either allow the user to close the app directly, or show the confirmation modal.

taahirisaacs, 2016-11-08:
Created a simple warning modal to alert the user if they close the application while an image is being flashed.

[image]

lurch, 2016-11-08:
Looks good. But I think "...sure you want to..." is more idiomatic than "...sure you would like to..."

WasabiFan, 2016-11-08:
It's worth noting that the implications of this action aren't immediately clear to the user. I think that adding a sentence explaining what will happen would be nice. Also, currently "what will happen" is that Etcher will continue flashing headless :laughing:

lurch, 2016-11-08:
Hmm, given that the dialog in #840 says "You will be able to overwrite your system drives if you're not careful.", maybe this dialog could also say something like "This may leave your drive in an unusable state." ?

> Also, currently "what will happen" is that Etcher will continue flashing headless :laughing:

Yeah, that's a bug #850

taahirisaacs, 2016-11-08:
Updated copy on design.

[image]

jviotti, 2016-11-09:
@taahirisaacs Looks good! Following @konmouz line of thoughts regarding these kinds of modals, I think we can remove the "Cancel" button and just present "Yes, continue". The user can dismiss the modal by clicking somewhere else or in the top right corner cross icon.

WasabiFan, 2016-11-09:
> I think we can remove the "Cancel" button and just present "Yes, continue".

No! Please, no! Having just a "dismiss" button on actions like this is detrimental for the user. I have yet to see an existing app by a notable company or group that doesn't have a "no" option on dialog boxes like this, and there's a good reason for that: "dismissing" the warning of imminent danger is *not* the same as aborting the action that caused it. In the other recent case of the "unsafe mode" dialog, there wasn't an immediate threat, so I didn't have a significant issue with it; but in this case, it really does matter.

Having the "dismiss" button is great and gives the user a quick and natural choice, but it doesn't work for all cases, especially when the user intentionally initiated the offending action.

konmouz, 2016-11-09:
@taahirisaacs we can just remove the 'X' on this modal. In general when there is a question we need to have 'yes and no' CTAs. On the contrary when you need to make a selection (like when you select a drive) there could be just a 'continue' CTA and an 'X' one to dismiss.

PR #868 (2016-12-05): feat(GUI): confirm before quitting etcher while writing
When Etcher is writing to a drive and the user closes the window,
currently there is no confirmation. This commit changes that and
the user has to confirm the exit.

Other changes include the warning modal taking a `rejectionLabel` option
that shows a reject button when provided.

Closes: #744
Change-Type: minor
Changelog-Entry: Confirm before user quits while writing.
closed 2016-12-05 by jviotti