balena-io/etcher #750: Have to unplug/plug my card reader after flashing each card.

Tryum, 2016-10-05:
- Etcher version: 1.0.0-beta.15
- Operating system and architecture: Win10 x64

I'm using an old USB2 transcend multicard reader. It seems like etcher ejects the whole µSD reader device instead of the card as I have to unplug/replug the card reader to make a new card available.

jviotti, 2016-10-05:
Hey there,

Etcher ejects drives upon completion by default, but you can disable such feature by un-ticking "Unmount on success" on the settings screen (the cog icon in the top right corner of the application).

Regarding unmounting the SDCard but not the reader, I don't think this can be accomplished, since they both represent the same physical drive, but please re-open if you find a way to do so!

lurch, 2016-10-06:
Different USB card readers also work in different ways too... some readers appear as 'empty drives' when no card is inserted, and others only appear to the host OS _after_ a card has been inserted. Some readers with two slots (e.g. a full-size and micro SD) appear as two separate drives (which can be written independently), others only appear as a single drive (i.e. only one slot can be used at once).
So it's quite believable that some card readers will actually need to be physically unplugged, whereas others will only need to have a different card inserted.

@jviotti Should the "Unmount on success" option be renamed to "Eject on success" if that's what it's actually doing? (I dunno about other OSes, but on Linux 'unmount' and 'eject' are different things - a partition that's been unmounted can later be remounted, but a disk that's been ejected is non-functional until it's been removed and reinserted)

jviotti, 2016-10-07:
> @jviotti Should the "Unmount on success" option be renamed to "Eject on success" if that's what it's actually doing? (I dunno about other OSes, but on Linux 'unmount' and 'eject' are different [...]

In Windows, "unmounting" basically means "ejecting". In UNIX based systems, unmounted drives are still accessible through direct I/O, however in Windows, if you "unmount" a drive, you have to re-plug or "mount" again to be able to access it.

WasabiFan, 2016-10-07:
It's worth noting that the term "eject" is likely going to be understood by Windows users, while "unmount" probably won't have any significance to the majority of them.

jviotti, 2016-10-07:
Makes a lot of sense. Lets change the wording to "Eject" when running on Windows.

Tryum, 2016-10-07:
Concerning my initial issue, I was reporting this because the behaviour of the ejecting/unmounting from etcher seems different than the ejecting from windows explorer.

If there's anything I can do like spying with a debug tool or else, let me know ;)

jviotti, 2016-10-09:
Hi @Tryum,

We're using `RemoveDrive.exe` to handle this (see http://www.uwe-sieber.de/drivetools_e.html). If you could download the tool and experiment with its options that would be amazing. If you find a combination that leads to the results you're expecting, happy to make such change in Etcher!

jviotti, 2016-10-09:
I'll re-open this so we can investigate together!

Beatnukem, 2016-12-08:
Just wanted to mention: I am using a 2013 MacBook with Windows 10 (bootcamp). Using Etcher with the unmout setting enabled will 'remove' the card reader completely forcing you to reboot the system to get it back. Not sure, but maybe the unmount option should be disabled by default until this is fixed? You could also present a dialog at the end "Done, do you want to eject? [yes] / [no]".

alexandrosm, 2016-12-08:
ooh, that sounds annoying indeed. maybe we should hold off from ejecting
until that is either fixed, or at least we can divide the cases where we're
sure we're doing the right thing and those where we're not.

lurch, 2016-12-08:
@Beatnukem I assume this is the internal card-reader slot on the laptop you're talking about?
Can you try doing the same thing in Etcher under OSX, to see if it exhibits the same behaviour? (or whether it's just a Windows-specific thing)

jviotti, 2016-12-08:
@Beatnukem 

> Using Etcher with the unmout setting enabled will 'remove' the card reader completely forcing you to reboot the system to get it back. Not sure, but maybe the unmount option should be disabled by default until this is fixed? 

In the Windows world, "unmounting" means "ejecting", and there is no way to unmount a drive like in the UNIX world while still be able to open a file descriptor to it (unless you remove the drive letters from it, but that's a mess, since you'll have to manually assign it back in order to use the drive, since not even a reboot will work).

Since this is an internal SDCard reader, there is indeed no way to plug it back :/ 

This is the tool we use to eject drives on Windows: http://www.uwe-sieber.de/drivetools_e.html. Maybe you can play around with the options it provides to see if you can get it to do the right thing? 

```
RemoveDrive DriveSpec: [-v][-d][-l][-t][-h][-a][-b][-i][-s][-w:nnnn]

DriveSpec  is the drive to remove, e.g. U: or an NTFS mountpoint as
           "C:\CardReader\Multi Reader SD"
           or . for the current drive
           or \ for the drive of the RemoveDrive.exe
           or a volume name like \\?\Volume{433619ed-c6ea-11d9-a3b2-806d6172696f}
           or a kernel name like \Device\HarddiskVolume2 (Volume)
           or a kernel name like \Device\Harddisk2\DR0 (Disk)
           or a partition name like \Device\Harddisk2\Partition1
           or a device ID
           or a friendly name like 'Corsair Voyager'
           or a volume label like 'Backup01'
           Wildcards can be used.
[-47]      perform safe removal resulting in problem code 47
[-v]       remove the volume only instead of the drive or whole device
[-d]       remove the drive only instead of the whole device
[-L]       loop until success
[-e]       try to dismount and eject if the removal fails
[-t]       release TrueCrypt volumes hosted by the drive to remove
[-h]       show open handles (admin rights required, exprimental, may freeze!)
[-h:X]     show open handles on a different drive (e.g. a TrueCrypt
           volume whose container is on the drive to remove, not required
           if -t -h is used)
[-a]       activates Windows of applications owning the open handles, requires -h
[-w:nnnn]  wait nnnn milliseconds before closing the program (time to read its output)
[-s]       self delete removedrive.exe
[-b]       let Windows show the "Safe To Remove Hardware" balloon tip
           (may lead to long delays)
[-i]       stop Windows indexing service (CiSvc) for a moment if required (admins)
[-dbg]     show debug information
```

lurch, 2016-12-08:
I assume *other* internal SD card readers on Windows laptops don't have this same problem too? I wonder what the difference is here? Hardware-level problem, or Windows driver-level problem? (i.e. I wonder if a different driver might fix this?)

Tryum, 2016-12-08:
Well I guess it's exactly the same problem that I have, hower my reader
being external and USB, I can disconnect/reconnect it to get it back
without rebooting.
I think there is a difference between ejecting the volume and ejecting the
device, and etcher seems to eject the whole device.

Thibault Jochem
Software Engineer
EFREI Promo 2007

lurch, 2016-12-08:
I assume there'd be a **lot** more problem reports, if everyone who has a Windows laptop with an internal SD-reader who uses Etcher, had to reboot their PC before they were able to use their SD slot again!!

daturach, 2017-02-01:
Hello @jviotti, @lurch  

I have a similar issue with a internal multi-card reader that was reported in thread you mentioned here above (Error: Path must be a string. Received undefined). Basically, once the SD card is burned and ejected, there is no way to re-use the drives. A computer reboot is necessary.
In the meantime, I learned there is an option to not eject the card.

I am running Windows 7 French and here is the info I found about my reader:

From http://driverlibs.com/download/General-UDisk-USB-Device

HardwareIDs:
USBSTOR\DiskGeneric-Compact_Flash___1.01
USBSTOR\DiskGeneric-Compact_Flash___
USBSTOR\DiskGeneric-
USBSTOR\Generic-Compact_Flash___1
Generic-Compact_Flash___1
USBSTOR\GenDisk
GenDisk

Driver: 6.1.7601.19133

Hope that helps, if not let me know if you need more info.
Thanks

jviotti, 2017-02-01:
Hi @daturach ,

What happens when you plug an SDCard to your PC using such adapter? If you open My PC, do you see various "unassigned" drive letter apart from the one from your SD Card?

When you open Etcher, and open the drive selector modal, how many drive letters does it list for that drive? Are they correct, or is Etcher listing those "unassigned" letters?

Check my previous comment (https://github.com/resin-io/etcher/issues/750#issuecomment-265764658) about the CLI tool we're using to handle unmounting.

They have some particularly interesting options that it might be worth trying:

```
[-v]       remove the volume only instead of the drive or whole device
[-d]       remove the drive only instead of the whole device
```

If your SDCard is assigned to `E:`, you could try:

```sh
> RemoveDrive.exe E: -v
```

And:

```sh
> RemoveDrive.exe E: -d
```

To see if any makes a difference.

daturach, 2017-02-02:
Hello @jviotti 

SD card is connected to "h:"
[image]

A propos, could you make this dialog box resizable? :-)

Thanks for the tool. Will have a look later this week.

jviotti, 2017-02-02:
Hi @daturach ,

We have some designs to make this larger, and possible stop making it a modal in the future, which should resolve this problem. We'll share the designs very soon, and you can provide feedback if you like.

Thanks a lot, looking forward to hear how it went! In any case, I'm trying to purchase an adapter similar to the one you have, and see if I manage to reproduce here as well.

daturach, 2017-02-04:
Hello @jviotti 

Both tests required a PC reboot to get the H: drive back but they return different messages. 
Only H: is not available. I:, J: and K: are still there.

`removedrive h: -d`

RemoveDrive V2.8.4 (x64) - prepares drives for safe removal
Freeware by Uwe Sieber - www.uwe-sieber.de
Removing Drive 'Generic- SD/MMC USB Device' (H:)
success  (reactivation is possible)

`removedrive h: -v`

RemoveDrive V2.8.4 (x64) - prepares drives for safe removal
Freeware by Uwe Sieber - www.uwe-sieber.de
Removing Volume H:
success

lurch, 2017-02-06:
@daturach  For both tests, after running `removedrive` I assume you tried physically removing and then re-inserting the SD card a couple of times?
Does the "safely remove" option in the Windows systray also exhibit this "drive becomes unusable until you reboot" behaviour, or does it do something different?

@jviotti This [search](https://www.google.co.uk/search?q=Alcor+Micro+Corp+Card+Reader) suggests that Alcor Card Readers need a specific software driver installed, which suggests they behave differently from "standard" USB card readers (which don't require any additional drivers to be installed), so maybe that's why `RemoveDrive.exe` is giving us this "weird" behaviour for these really old USB card readers? (and if my assumption is correct and it only affects old card readers, I guess that would explain why this problem is so 'rare')

jviotti, 2017-02-06:
@daturach Thanks a lot for the testing! So to make it clearer, both with `-d` and `-v` only the correct drive was ejected but not the others from the same reader, right (which is the expected behaviour)?

> Both tests required a PC reboot to get the H: drive back 

That's odd. I wonder if passing `-47` along with these options result in the correct behaviour:

```
[-47]      perform safe removal resulting in problem code 47
```

So `-47 -v` would perform a "safe removal" of the "volume only".

@lurch Could be. I'd be interred to know how the "safely remove" option in the systray compares to the behaviour we see on `RemoveDrive.exe`.

daturach, 2017-02-06:
@lurch: Yep, I did unplug the card.
@jviotti: yes, only the correct drive.

Will verify later if `[-47]` does the correct job.

jviotti, 2017-02-06:
OK, its at least an improvement. I'll wait to heard how `-47` went before sending a PR.

daturach, 2017-02-06:
Ok, here we go:

`removedrive h: -47 -v`

RemoveDrive V2.8.4 (x64) - prepares drives for safe removal
Freeware by Uwe Sieber - www.uwe-sieber.de
Removing Volume H:
failed

`removedrive h: -47 -d`

RemoveDrive V2.8.4 (x64) - prepares drives for safe removal
Freeware by Uwe Sieber - www.uwe-sieber.de
Removing Drive 'Generic- SD/MMC USB Device' (H:)

Veto=IllegalDeviceRequest
failed

Ok, tomorrow, I am going to buy a new computer...Just joking :-)

jviotti, 2017-02-06:
Haha, that sucks. I wonder why that fails. I remember I got it running on Windows 10 at some points. I just got Windows 7, so I'll see if this is a Windows 7 only issue.

lurch, 2017-02-07:
@daturach Just to re-request - Does the "safely remove" option in the Windows systray also exhibit this "drive becomes unusable until you reboot" behaviour, or does it do something different?
(i.e. how do you normally "eject" a card from your SD reader, without causing the whole drive to disappear?)

daturach, 2017-02-07:
@lurch To answer your question, I do a right-click on the icon on the systray and then eject.

[image]

[image]

[image]

wellsi, 2017-02-12:
I have the same issue on a fairly new Lenovo desktop with built in card reader. Lenovo K450e (Model 10181). I was not able to find a set of RemoveDrive options that work. Eject from the windows menu works just fine.

WasabiFan, 2017-02-12:
@jviotti doesn't `diskpart` have functionality to eject drives? Is there a reason you can't use that instead of `RemoveDrive.exe`?

jviotti, 2017-02-12:
> @jviotti doesn't diskpart have functionality to eject drives? Is there a reason you can't use that instead of RemoveDrive.exe?

I don't know, but looks like something worth investigating. I'll do so and report back soon.

jviotti, 2017-02-12:
I found this: http://blog.mundy.co/kb/28-windows/505-safely-remove-a-disk-using-the-command-line

> Type "select volume <number>", where <number> is the number of your USB storage.
> Then, type "remove all dismount". Type "exit" to leave diskpart. Enjoy! ;)

However this removes the associated drive letter of the drive, meaning that you won't get Windows to recognize it back when you plug it again unless you manually assign a new drive letter to it (so its not really the same as the "Safely Remove" Windows feature).

I Googled for alternatives and found the following pages:

- http://superuser.com/questions/443162/remove-usb-device-from-command-line
- http://stackoverflow.com/questions/22705830/safely-remove-usb-from-batch-file

They all seem to recommend the RemoveDrive.exe file we're already using :(

lurch, 2017-02-13:
The same [page](http://www.uwe-sieber.de/drivetools_e.html) which hosts the `RemoveDrive` utility, also hosts an `EjectMedia` utility which it describes as "Ejecting the media only is especially suitable for internal cardreaders".
Is that something you've experimented with yet @jviotti ? (although if that _does_ work, I obviously have no idea how you'd know when to use `RemoveDrive` and when to use `EjectMedia`!)

Looks like [this](http://www.uwe-sieber.de/usbstick_e.html) might have some useful info too...

jviotti, 2017-02-17:
Check this: https://www.codeproject.com/articles/259577/how-to-flush-a-storage-volumes-the-file-cache-lock. A good explanation of how `EjectMedia` works. It looks like we have enough technical information to go and implement our own unmounting program.

jviotti, 2017-02-17:
OK, I think I found something that works. @daturach @Tryum Can you try the [EjectMedia](http://www.uwe-sieber.de/files/ejectmedia.zip) tool calling it like this?

```sh
EjectMedia.exe X: -f
```

Obviously replacing `X:` with your corresponding drive letter.

I've tested this on the internal reader of a MacBook Pro running BootCamp, on an USB Hub, on

jviotti, 2017-02-17:
Sorry for the open/close. I've hit the wrong button :)

jviotti, 2017-02-17:
Not passing the `-f` option causes the internal reader to remain active on My PC even though its empty.

daturach, 2017-02-20:
Ok @jviotti . I got the same results as you and I don't need to reboot anymore! Well done
```
EjectMedia.exe H: -f

EjectMedia V2.7.2 (x64) - ejects a media from a drive
Freeware by Uwe Sieber - www.uwe-sieber.de
Ejecting media from H: (Generic- SD/MMC USB Device)
Flushing cache... OK
Allowing eject... OK
Locking volume... OK
Dismounting volume... OK
Ejecting media... OK
success
```
Will EjectMedia work with any drives ? Internal or external?

Without the -f option, same as above:
```
EjectMedia.exe H:

EjectMedia V2.7.2 (x64) - ejects a media from a drive
Freeware by Uwe Sieber - www.uwe-sieber.de
Ejecting media from H: (Generic- SD/MMC USB Device)
Flushing cache... OK
Allowing eject... OK
Locking volume... OK
Dismounting volume... OK
Ejecting media... OK
success
```

jviotti, 2017-02-20:
Amazing news @daturach ! As far as my experimentations went, this new tool works fine for both internal and external drives. Weird that omitting `-f` yields the same result to you. I'll keep it just in case anyway.

I'll be sending a PR to make use of this tool very soon.

jviotti, 2017-02-20:
Just for reference, I double checked that we're allowed to re-distribute this:

From http://www.uwe-sieber.de/drivetools_e.html

> Allowed:
> - usage in any environment, including commercial
> - include in software products, including commercial
> - include on CD/DVD of computer magazines

Since we know how this tool works, we might re-implement it and extend it ourselves sometime in the future anyway.

@jhermsmeier ^

lurch, 2017-02-20:
So just to clarify, are you saying that in hindsight we should have just used `EjectMedia.exe` instead of `RemoveDrive.exe` all along? ;)

If you haven't already done so, it might be worth checking to see what EjectMedia does when you try to use it on a USB HDD as opposed to a USB flash drive?

jviotti, 2017-02-20:
Looks like you're right. EjectMedia works on USB drives but not on USB HDD (it does nothing in that case). I think we can try EjectMedia, and if the child process exits with code 0 but still drivelist detects that the drive is still mounted, then we fallback to RemoveDrive. What do you think?

lurch, 2017-02-20:
Sounds reasonable, if sufficient testing shows that it works as expected? Would we need to insert some kind of delay between running EjectMedia and re-scanning the driveslist in case it takes a couple of seconds for the drive to unmount / disappear? (or is that guaranteed to have happened by the time EjectMedia exits?)

And that raises an interesting side-question - as well as returning the info for _all_ drives, would it be worth adding a mode to drivelist to return information only for a specific drive? (in case that's quicker than scanning all drives)

jviotti, 2017-02-20:
> And that raises an interesting side-question - as well as returning the info for all drives, would it be worth adding a mode to drivelist to return information only for a specific drive? (in case that's quicker than scanning all drives)

This is an amazing idea. Updating the scripts to handle both a single drive or all drives might be tricky, but we can start by exposing a function to get information about a single one that simply runs the script and filters for the one you're interested in, and optimize later.

closed 2017-04-03 by jviotti