balena-io/etcher #756: Error: No polkit authentication agent found.

TheSandman86, 2016-10-12:
- **Etcher version: 1.0.0-beta.15**
- **Operating system and architecture: Xubuntu 16.04.1 64 bits**
- **Do you see any meaningful error information on DevTools?**

I'm getting this error while triying to flash an image on a SD Card,
This is what i see on the DevTools console:
Wed Oct 12 2016 11:56:07 GMT-0300 (ART) Flash error
tracker.js:10 Wed Oct 12 2016 11:56:29 GMT-0300 (ART) Flash ({"image":"/home/mauricio/Downloads/ubuntu-16.04.1-desktop-amd64.iso","device":"/dev/mmcblk0"})
tracker.js:10 WRITER: Tailing /tmp/etcher-2553487Od3NFKK75z.log
tracker.js:10 WRITER: Tail starting:
tracker.js:10 WRITER: filename: /tmp/etcher-2553487Od3NFKK75z.log
tracker.js:10 WRITER: options: {}
tracker.js:10 WRITER: Attempting to elevate
tracker.js:10 Error: { Error: No polkit authentication agent found.
    at /tmp/.mount_7QwaJI/usr/bin/resources/app.asar/node_modules/sudo-prompt/index.js:132:23
    at ChildProcess.exithandler (child_process.js:221:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:850:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)
  cause: 
   Error: No polkit authentication agent found.
       at /tmp/.mount_7QwaJI/usr/bin/resources/app.asar/node_modules/sudo-prompt/index.js:132:23
       at ChildProcess.exithandler (child_process.js:221:5)
       at emitTwo (events.js:106:13)
       at ChildProcess.emit (events.js:191:7)
       at maybeClose (internal/child_process.js:850:16)
       at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5),
  isOperational: true }

    at Socket.child.stderr.on (/tmp/.mount_7QwaJI/usr/bin/resources/app.asar/lib/src/child-writer/index.js:83:29)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:172:18)
    at Socket.Readable.push (_stream_readable.js:130:10)
    at Pipe.onread (net.js:542:20)
a.(anonymous function)	@	tracker.js:10
(anonymous function)	@	/tmp/.mount_7QwaJI/usr/bin/resources/app.asar/node_modules/angular/angular.js:13708
logException	@	/tmp/.mount_7QwaJI/usr/bin/resources/app.asar/lib/gui/modules/analytics.js:157
reportException	@	/tmp/.mount_7QwaJI/usr/bin/resources/app.asar/lib/gui/modules/error.js:45
ImageWriterService.flash.then.catch	@	/tmp/.mount_7QwaJI/usr/bin/resources/app.asar/lib/gui/pages/main/controllers/flash.js:82
processQueue	@	/tmp/.mount_7QwaJI/usr/bin/resources/app.asar/node_modules/angular/angular.js:16170
(anonymous function)	@	/tmp/.mount_7QwaJI/usr/bin/resources/app.asar/node_modules/angular/angular.js:16186
$eval	@	/tmp/.mount_7QwaJI/usr/bin/resources/app.asar/node_modules/angular/angular.js:17444
$digest	@	/tmp/.mount_7QwaJI/usr/bin/resources/app.asar/node_modules/angular/angular.js:17257
(anonymous function)	@	/tmp/.mount_7QwaJI/usr/bin/resources/app.asar/node_modules/angular/angular.js:17483
completeOutstandingRequest	@	/tmp/.mount_7QwaJI/usr/bin/resources/app.asar/node_modules/angular/angular.js:5955
(anonymous function)	@	/tmp/.mount_7QwaJI/usr/bin/resources/app.asar/node_modules/angular/angular.js:6234
d.util.hasFunction.f.(anonymous function)	@	tracker.js:3

This error happens since version 1.0.0-beta.14 on my Thinkpad T440 and T420, but the last version now works on the T420, both of them running Xubuntu 16.04.1 64 bits.

jviotti, 2016-10-12:
Hi @TheSandman86 ,

Etcher requires an available [polkit authentication agent](https://wiki.archlinux.org/index.php/Polkit#Authentication_agents) in your system in order to show a secure password prompt dialog to perform elevation. If you're running Xubuntu, you might be interested in [xfce-polkit](https://github.com/ncopa/xfce-polkit).

I'll add the to the docs in any case.

TheSandman86, 2016-10-12:
Hi, thank you for the fast response,  i've installed the xfce-polkit and now Etcher works without errors. Is there any way to bypass this requeriment? or to make it optative?
Thanks again

jviotti, 2016-10-13:
We could bypass this by implementing our own authentication dialog, however this is not desirable since from the user point of view, we could be doing anything with the password. Doing elevation through polkit means that our users can trust us that we don't handle the password in plain text at any point.

Notice we are working on providing deb/rpm packages, which will list a polkit agent in their dependency list, and thus will ensure one its available.

TheSandman86, 2016-10-13:
Great, thanks for the info. I will test it when its released. This issue is solved for me, i don't know if you want to keep it open. Do i close it?