balena-io/etcher #759: ISO flash ok but it is not correctly

DiegoAccorinti, 2016-10-13:
- **Etcher version:**   v1.0.0-beta.15
- **Operating system and architecture:**  Debian Jessie AMD64

When trying to flash an ISO image, the procedure ends OK but seeing the results in Gparted are:

[image]

ISO itself flashed with unetbootin is as follows:

[image]

why it is marked as "hidden"? and why the image does not occupy the entire space as it does with netbooting?
In addition, it clarified that the pen drive fails when trying to boot the ISO.
regards.

lurch, 2016-10-13:
See https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos and https://github.com/resin-io/etcher/search?q=unetbootin for more info.

DiegoAccorinti, 2016-10-13:
tnks,  but dont solute the problem. The pen don't boots.

DiegoAccorinti, 2016-10-13:
ok   the pendrive boots OK in a non-EFI PC.   tnks

closed 2016-10-13 by DiegoAccorinti