balena-io/etcher #760: no portable version

tooomm, 2016-10-15:
- **Operating system and architecture:** Win 10 64bit

I would highly appreciate to see a portable version of this sweet tool! ❤️

jviotti, 2016-10-15:
Hi @tooomm,

Thanks for the suggestion! We do build a portable version at one point during the packaging, but only re-distribute the final installer. Uploading the portable version as well sounds like a good idea!

tooomm, 2016-10-16:
You already somehow build it? That sounds great.
Makes just distributing it an easier task. :)

jviotti, 2016-10-17:
The installer gets created around an standalone version of the app. Of course, the standalone version consists of many files (dlls and other resources) but can be run from anywhere. 

Is this what you're looking for, or do you mean a single standalone executable instead?

lurch, 2016-10-17:
I'm assuming he meant something along the lines of http://portableapps.com/ i.e. that can just be run by extracting a zipfile to a local folder?

tooomm, 2016-10-17:
Actually 
>many files (dlls and other resources) but can be run from anywhere

is pretty much what i'm looking for. 
Is there are a version hosted somehwere that i can test?

I don't use suits like portableapps.com, having just a single executable file is easier to handle for sure!

jviotti, 2016-10-18:
@tooomm

Sounds good! We don't have builds like that uploaded anywhere, but you can build one from source if you like (see https://github.com/resin-io/etcher/blob/master/docs/RUNNING-LOCALLY.md. Once you have a locally running build, you can run `.\scripts\build\windows.bat package <arch>` to get the portable build).

tooomm, 2016-10-18:
I happily wait for a official distribution then. ;)

DJuego, 2016-10-19:
Me too. 👍  Please! :-) 
Thanks!
DJuego

P.S.  I have discovered this tool from ev3dev project.

closed 2016-10-25 by jviotti

lurch, 2016-11-03:
Just in case the people who've commented on this issue haven't seen this yet:
https://talk.resin.io/t/etcher-1-0-0-beta16-release/379