balena-io/etcher #761: Older version?

H1ghTech, 2016-10-16:
Hi

Do you have older versions for os x 10.7.5 ?

Thanks in advance
H1ghTech

jviotti, 2016-10-17:
Hey there! Thanks for reaching out. Sadly, [electron](http://electron.atom.io), the framework we use to build the graphical user interface, only supports 10.9 (see https://github.com/electron/electron/blob/master/docs/tutorial/supported-platforms.md).

You may want to check out the CLI version of Etcher, which should run fine on that platform. See https://github.com/resin-io/etcher/blob/master/docs/CLI.md.

jviotti, 2016-10-17:
I added a note to myself to put this in the docs.

closed 2016-10-17 by jviotti

berryamin, 2019-07-06:
Ethcher-cli was apparently removed.
Don't know the reason behind this, but I need ethcher-cli to install on a 10.7.x

You may do a "git checkout 4568404a704ac367824c600d368c10cd0516f4a9" to have it re-appear in your repo .. Will not have the latest patches however ..

From da072e7621fd57e09d6a60c429f17a4d24c8cd72 Mon Sep 17 00:00:00 2001
From: Alexis Svinartchouk <alexis@resin.io>
Date: Thu, 14 Feb 2019 16:40:40 +0100
Subject: [PATCH 1656/1806] Remove etcher-cli

berryamin, 2019-07-06:
Latest released binary file for Etcher cli for Darwin is 1.4.9 :
https://github.com/balena-io/etcher/releases/tag/v1.4.9

cr2007, 2021-03-25:
> Hey there! Thanks for reaching out. Sadly, [electron](http://electron.atom.io), the framework we use to build the graphical user interface, only supports 10.9 (see [...]

@jviotti 

Hey there!

I have a MacBook that has OS X 10.9.5
Since you mentioned the final GUI-based one supports 10.9
Is there a way we can download that version?

Would be really helpful just to put my Linux Distribution onto a flash drive

lurch, 2021-03-25:
@cr2007 **All** the old Etcher versions are available at https://github.com/balena-io/etcher/releases but I've no idea how you'd figure out which was the last version to support OS X 10.9.5 ?
Perhaps you can get some hints from https://github.com/balena-io/etcher/blob/master/CHANGELOG.md  :shrug:

cr2007, 2021-03-25:
@lurch 
It's alright
I found another way to download it to put the Linux distro to my flash drive.

Thanks! 😄