balena-io/etcher #762: Feature request: pre-populated list of images a-la UNetbootin

bomberstudios, 2016-10-18:
Probably needs #54 to happen first, but I thought I'd ask. 

Currently I use UNetbootin for my flashing needs, and one of the nice features it has is that it includes a list of images by default, so flashing a new distro is as easy as choosing it from a drop down and clicking "Burn".

Would something like this be useful in the use cases you have in mind for Etcher? I can definitely see myself using the feature, but I'm not 100% I'm part of the target market for the app : )

jviotti, 2016-10-18:
Hi @bomberstudios,

Thanks for reaching out! This is definitely something we want to build in Etcher. I'll close this ticket in favor of https://github.com/resin-io/etcher/issues/740, which discussed the possibility of publishers to distribute "metadata" extended images which basically contain a set of URLs that would allow you to choose from different options.

For example, Canonical could provide an "Ubuntu extended archive" which contains URLs to all officially supported Ubuntu flavors and versions, which would cause Etcher to render a dropdown like UNetBootin.

closed 2016-10-18 by jviotti

bomberstudios, 2016-10-18:
Awesome, thanks for the explanation : )

lurch, 2016-10-18:
AFAIK UNetBootin is targeted primarily at writing LinuxLive images to USB sticks, for x86 / x86-64 PCs. Whereas Etcher has a _much_ broader target than that (both in terms of distros, and target platforms), so any "built in list" like unetbootin would never be able to meet everyone's demands.

jviotti, 2016-10-19:
Definitely. Etcher won't have a built-in list but will accept metadata extended archives which provide a list that fills a specific niche, which is not exactly what the OP asked for, but somewhat close to that, while generic enough for Etcher.