balena-io/etcher #763: Name installer executables (and portable versions too) so that they include the Etcher version number

WasabiFan, 2016-10-18:
Currently, my Downloads folder looks like this:

[image]

You'll notice that, aside from the debug builds I downloaded for diagnosing an error, I have a bunch of Etcher installer executables which are auto-incrementing with a suffixed number as I download new ones. For me at least, I have had to (and likely will have to in the future) go back to old versions of Etcher for various reasons, and attempting to discern which version is which is pretty much impossible with it like that. _Yes, I could name them myself or delete old versions, but I'm a lazy programmer who ain't got time for that._

I think it would be helpful if Etcher installers were named such that they included the version number of the app that they contained. Not only would this prevent name collisions (and by extension auto-incremented names) but it would also make it much easier to figure out which version you downloaded a week ago. It would be even nicer if, for the case where one is running special builds, the installer name included the user-supplied name of the person who built the thing (which would be set as part of the build process).

lurch, 2016-10-19:
On the other hand, see #336

WasabiFan, 2016-10-19:
The update prompt just links the user to the download button on the Etcher website, so that doesn't help this issue. Even if Etcher implemented the download and reinstallation itself, I would still like the version number to be in the name.

alexandrosm, 2016-10-19:
I like the idea too, for what it's worth. I think it's just a matter of
doing it.

jviotti, 2016-10-20:
Makes a lot of sense. Let's go for it!

jviotti, 2016-10-27:
This issue has been fixed for all operating systems by https://github.com/resin-io/etcher/pull/764

closed 2016-10-27 by jviotti