balena-io/etcher #773: Elaborate Error when writing to disk. What is wrong?

mendeldesign, 2016-10-25:
[image]
[image]

- **Etcher version: 1.0.0-beta.15**
- **Operating system and architecture: OSX 10.11.6**
- **Do you see any meaningful error information on DevTools?**

mendeldesign, 2016-10-25:
It seems to be saying that the image file was not ok, but that img file actually worked a month ago on a different SD card.

lurch, 2016-10-25:
Looks like it's going wrong because of the `&` in the file-path? See also #738

mendeldesign, 2016-10-25:
O wow you are right, dragged it to the desktop and it works. Thank you for the quick reply, I was already downloading new images from the raspberry website.

closed 2016-10-27 by jviotti