balena-io/etcher #778: [feature] Give more info about and/or visualise the writing process

alexandrosm, 2016-10-26:
I put this down as a very high-level feature discussion, but it would be quite nice to have the ability to see in detail the progress of the writing process. Many technical users prefer dd to etcher, but if they had access to yummy data as the process was ongoing, I suspect we may tempt them at the least. Also, the more you see happenning the faster the time seems to go, so it would help with the perceived speed. And of course, if we can encourage less technical users to think about the process, maybe we'll spark some interest :).

I'm not sure if this should be the default view, or some kind of "advanced" view, but it would be nice to have one way or another. One more meta-idea is that it would be fun to also have the resulting data accessible as a .json file if the user wants to see them?

In terms of concrete ideas, I remember in the old days we had scandisk on ms-dos that looked something like this:
https://i.ytimg.com/vi/YdKB1UzDC5M/hqdefault.jpg

It would be nice if we could do something similar on etcher.

As about the "what data" to visualise, if we follow the image above, we could think about things like the location of bad blocks (which has been previously discussed), what has been written and what has been verified, and maybe potentially what "will" be written (vs what will be left blank). Perhaps the bmaps also offer some interesting data about which blocks are going to be left empty and not.

Not sure if we can get access to the data, but if we can, it would also be nice to display the boundaries between partitions etc.

Another type of data is the speed in which the write is proceeding in some kind of "speed over time" chart.

Any other ideas about relevant data and visualisations that it might be cool to show? Any opinions about whether the feature as a whole makes sense?

lurch, 2016-10-27:
Definitely a good idea :-)
We could perhaps also estimate the time to completion? (but if we do, it'd obviously be better to over-estimate than under-estimate)

WasabiFan, 2016-10-27:
Doesn't Etcher already estimate the time?

lurch, 2016-10-27:
Ummm, oops ;)

jviotti, 2016-10-27:
This would be really, really cool! Maybe we can loop @taahirisaacs to provide some sketches of things he can think of, as a way to continue discussion?

As a side note, Apple released a [new MacBook Pro](http://www.apple.com/macbook-pro/) today, which has this nice "Touch Bar" feature. I wonder how this idea would apply there :)

lurch, 2016-10-27:
This also touches slightly on some of the stuff we previously discussed in #735

alexandrosm, 2016-10-27:
Yeah. I had had similar ideas but that thread kinda rekindled/meshed with
them

jviotti, 2018-10-15:
I'm going to close this as this already landed in Etcher v2 design

closed 2018-10-15 by jviotti