balena-io/etcher #781: AppImages README

lurch, 2016-10-27:
As Etcher is an OpenSource project, it might be nice if there was a little README in https://github.com/resin-io/etcher/tree/master/scripts/build/AppImages explaining where those binary files came from ( https://github.com/probonopd/AppImageKit ? ), which git commit ID they were pulled from, how they were compiled, etc. etc.?

jviotti, 2016-10-27:
Sounds great, thanks a lot for the heads up on this. I'll sketch a document probably tomorrow. As a side note, most of these things don't need to live in the repository, so I guess we can put code in the build scripts to fetch them and thus make a big part of the documentation unnecessary.

lurch, 2016-10-27:
Great, eliminating the number of binaries stored in source-control is always a good thing :-)

lurch, 2016-10-28:
BTW: https://github.com/probonopd/AppImageKit/pull/120#issuecomment-256837141
(but I've obviously got no idea for the timescale on appimagetool being production-ready)

jviotti, 2016-10-28:
Very interesting! I suggest postponing the upgrade to the new version until it becomes stable. We're pushing AppImages to their limits to handle elevation, and would hate to do lots of work to update to the new format, and have the new format dramatically change directions, and have us duplicating work.

lurch, 2016-10-28:
> I suggest postponing the upgrade to the new version until it becomes stable.

I totally agree :+1:

Probably wouldn't hurt to set up a 'private experimental branch' testing out the basics of the appimagetool, in case it fundamentally doesn't work for Etcher's use-case without additional feature requests? It would suck for Etcher to be suck with the "no longer developed" AppImageAssistant. Maybe that's something I should have a look at once I've done the packaging-script-refactoring?

Probably not worth changing the files in https://github.com/resin-io/etcher/tree/master/scripts/build/AppImages until we've made this AppImageAssistant / appimagetool assessment?

jviotti, 2016-10-28:
> Probably wouldn't hurt to set up a 'private experimental branch'
> testing out the basics of the appimagetool, in case it fundamentally
> doesn't work for Etcher's use-case without additional feature [...]

Yeah, definitely! I'll give it a go some time during next week if time
permits, and will report back how it went.

> Probably not worth changing the files in
> https://github.com/resin-io/etcher/tree/master/scripts/build/AppImages
> until we've made this AppImageAssistant / appimagetool assessment?

Sounds good to me :+1:

probonopd, 2016-10-28:
> We're pushing AppImages to their limits to handle elevation, and would hate to do lots of work to update to the new format, and have the new format dramatically change directions, and have us duplicating work.

Don't worry, the format is _not_ going to dramatically change directions or require major changes on your side, but will come with additional functionality such as:

* Binary delta update information can now be embedded even easier
* GPG2 signatures can be embedded inside AppImages
* Both can be done automatically for you by `appimagetool`
* Under the hood, we are using squashfs rather than ISO9660 now but you should not notice the change
* AppImages can extract themselves easily (e.g., in cases where `/dev/fuse` permissions are not available to a user)
* An optional daemon, `appimaged`, can handle desktop integration such as registering menus, icons, MIME types, etc. with the system

jviotti, 2016-10-28:
Great to know, thanks a lot for the clarification @probonopd. I'll be
trying this out very soon.

jviotti, 2016-11-28:
The binaries are now being downloaded from the internet, so I guess this issue is good to close. I'll be upgrading to `appimagetool` very soon.

closed 2016-11-28 by jviotti