balena-io/etcher #785: Website download instructions advise users to attempt to execute a zip file as an executable.

emctague, 2016-10-29:
- **Etcher version:** 1.0.0
- **Operating system and architecture:** linux x64
- **Do you see any meaningful error information on DevTools?** no

It seems that the installation instructions for the linux version of etcher are broken. It currently advises the user to `chmod +x` the downloaded zip file, instead of extracting it and performing operations on the AppImage it contains.

[image]

It also seems to give the same advice for the windows download:

[image]

jviotti, 2016-10-29:
Thanks for the heads up! Fixed by https://github.com/resin-io/etcher-homepage/commit/3335846b447d0b86b0cf2c1cb0185605d09a05c7

closed 2016-10-29 by jviotti

jviotti, 2016-10-29:
I've opened a new ticket for the fact that the GNU/Linux instructions are being shown for Windows: https://github.com/resin-io/etcher-homepage/issues/20