balena-io/etcher #787: Way of displaying supported formats

tooomm, 2016-10-29:
- **Etcher version:** 1.0.0-beta.16 portable
- **Operating system and architecture:** win 10 64bit

Why do you hide the most common image format `iso` behind "many more"?
Every bios or fw update, and all os images for linux and win I run into use it...
[image]
Shouldn't it be the other way around?
Show common ones that people know and use most likely, and append more specific and advanced ones?

Maybe there should even be some ordering. Show some important ones, others go to the tool tip in alphabetical order for example. Right now you get a bit lost in the tooltip. :)

jviotti, 2016-10-31:
@tooomm Thanks for the feedback. This is a great point. Currently we just take the first 3 formats defined by `etcher-image-stream` and put the rest in the tooltip.

Determining which ones should be presented outside the tooltip might be tricky, since for one side, we want highlight and make more obvious the more exotic formats we support, while still make clear that we support the "obvious" ones, like `.iso`.

I'd love to have input from more people on this one.

lurch, 2016-10-31:
What about "img, iso, zip, etch and many more" ?
`img` and `iso` are the image formats that people are most used to, `zip` shows that it also transparently supports compressed files, and `etch` shows that it also supports exotic formats ;-)

(AFAICT Etcher was originally developed for writing .img files to SD cards, and that's why IMHO `img` should be listed before `iso`)

Although that _might_ then make the text under the button wider than the button itself (?) and I dunno what @taahirisaacs thinks about that...

tooomm, 2016-10-31:
>Although that might then make the text under the button wider than the button itself (?) 

It could just say "and more" or "and others" instead. Maybe even just some "..." dots or the (?) help icon are enough.

alexandrosm, 2016-10-31:
I agree with Andrew's proposal, except that i don't see the reason to
promote .etch yet. For the moment it's a stub of a format we may make more
real in the future.

jviotti, 2016-10-31:
Sounds good to me. What do you think about @tooomm 's suggestion about changing the text afterwards? I kinda like "and many more", since it sounds like a more "excited" way of promoting if, but maybe that's just me.

alexandrosm, 2016-10-31:
"and much more" sounds better in some unspecified way?

jviotti, 2016-10-31:
Yeah, I like it. Lets go for it!

closed 2016-11-02 by jviotti