balena-io/etcher #792: Move / copy / add documentation to website?

lurch, 2016-10-31:
There's currently lots of docs in https://github.com/resin-io/etcher/tree/master/docs but very little on https://www.etcher.io/

It would be great to see more user-focused documentation on https://www.etcher.io/ as I doubt many users will bother clicking through to GitHub. Maybe this issue should be added to https://github.com/resin-io/etcher/milestone/2 ?

alexandrosm, 2016-10-31:
> etcher/tree/master/docs but very little on https://www.etcher.io/

But there is only one doc in there targetted at users though, no?

lurch, 2016-10-31:
IMHO https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md is bit of a mish-mash of FAQs, notes, and general info, and ought to be tidied up and/or split into sub-categories before being published as "end user documentation" (and not just "developer-level documentation") on https://www.etcher.io/

I appreciate that Etcher is supposed to be simple enough that it doesn't really _need_ documentation, but some people may be put off by the 'sparseness' of https://www.etcher.io/ ? The only real "documentation" currently is the animated GIF in the middle of the page.

All IMVHO, naturally :-)

Perhaps this issue should be moved to https://github.com/resin-io/etcher-homepage

alexandrosm, 2016-10-31:
> DOCUMENTATION.md is bit of a mish-mash of FAQs, notes, and general info,
> etcher-homepage

don't get me wrong, I agree that we should have docs. I also happen to
agree we should break up user-documentation.

The only reason you may be seeing this be stuck is that we're envisioning a
better way to do the etcher landing page altogether, which includes
auto-importing the docs from the repo, that is making us hesitant to add
much to the existing one.

It's a bad instinct, but that doesn't make it go away :)

lurch, 2016-10-31:
Haha, I've just been catching up with https://gitter.im/resin-io/etcher and see you were discussing this exact issue a few days ago! :laughing:

lurch, 2018-02-26:
This is a "Backlog" task, but pinging @zwalchuk in case it's something he wants to think about ;-)

closed 2018-10-15 by jviotti