balena-io/etcher #793: Rename "Standalone" version

imrehg, 2016-10-31:
There's a new set of "Standalone" Windows versions of Etcher to download. It was pretty confusing trying to figure out what it means. It would be great to mark that "Portable" for clarity, and possibly "Installer" the other version, for clarification.

jviotti, 2016-10-31:
I'll change "Standalone" to "Portable" and add "Installer" to the other ones as suggested. I think we should aim to obviousness rather than conciseness.

closed 2016-10-31 by jviotti

WasabiFan, 2016-10-31:
Alternately, you could make it two steps: First select your OS and then select the type. Or, maybe add OS headers in the list. As it is, it takes some real thinking to figure out which one you want.

lurch, 2016-10-31:
Yeah, having 7 different download options in a single drop-down menu might be a bit overwhelming for some people.

jviotti, 2016-11-01:
@lurch @WasabiFan I agree with all feedback. We should put more serious efforts in re-designing the site and add new improvements. /cc @craig-mulligan @taahirisaacs

alexandrosm, 2016-11-01:
I guess the obvious improvement is to have something like:

Download now!
[Windows V] - [x64 V] - [Installer V]

Where the V is the down arrow in a dropdown. Basically have sane defaults
and let people choose if they want to alter any individual dimension.

imrehg, 2016-11-01:
@alexandrosm please don't do multi-dropdown, IMHO that's truly horrendous :/

WasabiFan, 2016-11-01:
> please don't do multi-dropdown, IMHO that's truly horrendous :/

I think it's better than it is now, especially considering that the drop-downs would (presumably) default to your auto-detected OS. I can't think of any better options at the moment either.

lurch, 2016-11-02:
Given that Windows is the only platform with separate Portable / Installer builds, if you select e.g. Linux as the OS, would the third dropdown disappear? ;)

alexandrosm, 2016-11-02:
yeap, but keep in mind that there are other "package types" like "debian
package" we may want to highlight here.

And wait till we get to "GUI vs CLI" editions!

@imrehg I agree it can be done very badly, but I also suspect we can do it
decently well if we pay attention to it. It's just a lot more fiddly to get
right than other UIs, but I trust this crowd to smash it until it's perfect
:)

craig-mulligan, 2016-11-02:
@jviotti did you see the site revision yesterday? https://craig-mulligan.github.io/etcher-website/

It connects to #792 and only pulls the new downloads when the site is built. 

I've moved the downloads to a table, and then do an auto-detect os/arch and populate the main download button in the section. 

I quite like node.js download page: 

Where they highlight defaults and list other options: 
https://nodejs.org/en/download/current/

lurch, 2016-11-02:
> I quite like node.js download page:
Where they highlight defaults and list other options:
https://nodejs.org/en/download/current/

Side note: The thing I hate about the node.js download page is that they offer binary downloads for Linux, but **nowhere** (that I could find) do they give instructions on how to install them! :-(
(I ended up using https://nodejs.org/en/download/package-manager/#debian-and-ubuntu-based-linux-distributions instead)

WasabiFan, 2016-11-02:
@craig-mulligan One thing I think really needs improvement on that revision concept you linked to is the architecture column of the table. It's very hard to scan the list and figure out which one is the one you want. Could you write out "32-bit" and "64-bit" instead of the shortcut notations?

jviotti, 2016-11-02:
The revision looks very nice already, I like the table layout for the
downloads. I don't like the main action button text very much though.
"Try Etcher" feels like if we're offering a trial to users. Maybe
"Download Etcher"?

WasabiFan, 2016-11-02:
A few pedantic notes:
- Could you hyphenate "Hard drive friendly"? 
- Is the conclusion that "etcher" is capitalized or not? I think it would seem more logical (especially in a title) if it were capitalized.

Over all, I think it's a significant usability improvement over the existing site (which is the main reason I'm participating in these discussions). Nice job!

P.S. It would be really cool if the preview gif included the progress bar filling up :laughing:

dlech, 2016-11-02:
I like sites that offer you the most sensible download based on your browser info and then have a link for "more download options" that takes you to a downloads page that lists *all* of the available options with some helpful explanations.

alexandrosm, 2016-11-02:
Yeah, i think that would be ideal.. a sane default and a link to more
options. The table could be a bit more condensed and not lead with the long
filename maybe?

craig-mulligan, 2016-11-03:
Okay so in the header section we replace the 'Try etcher' with 'Download etcher for `<os>`' and then have a link below the takes them to the table of downloads. I think we should keep the downloads section on the homepage, I don't think it's necessary to add it's own page until we have more download options.

In the table we simplify the filename and add `portable/installer` where appropriate. Also, make `32-bit/64-bit` more prominent (bold?)

All in agreement?

lurch, 2016-11-03:
Could there also maybe be a "separator bar" or something between the downloads for the different OSes?