balena-io/etcher #797: README.md doesn't match "style guide"

lurch, 2016-10-31:
According to https://github.com/resin-io/docs#style-manual **resin.io** shouldn't be capitalised, whereas `README.md` in this repo currently says "the Resin.io team".
I won't submit a PR to fix this myself, because it's unclear whether "the Resin.io team" should be updated to "the resin.io team" or "the resin team".

lurch, 2016-10-31:
This is also present in `SUPPORT.md`

alexandrosm, 2016-10-31:
you got it right. feel free to PR. it should be all-lowercase.

lurch, 2016-10-31:
So, which is correct: "the resin.io team" or "the resin team" ?

And I guess all the docs should similarly be updated to lower-case 'etcher' instead of 'Etcher'?

alexandrosm, 2016-10-31:
the former. I'm not sure we have similar guidelines for Etcher, let's
discuss next week?

jviotti, 2016-10-31:
@alexandrosm The link @lurch shared also mentions that "Etcher" should be "etcher" instead. Lets do both changes I guess.

dlech, 2016-10-31:
Names stylized with a leading lower-case letter are annoying (I know first hand from "ev3dev"). Please don't use "etcher". (Unless you are also doing something else to stylize it, like making it bold or using a different color or adding &#8482; or &reg; or something like that.)

dlech, 2016-10-31:
Names that come from urls aren't so bad, so "resin.io" doesn't bother me so much.

jviotti, 2016-10-31:
Yeah, I agree with @dlech that "Etcher" looks *way* better than "etcher".

WasabiFan, 2016-10-31:
And it's a real word as well, so capitalizing it makes it clear that Etcher is a product.

lurch, 2016-10-31:
Ping @sonyagreen and @shaunmulligan  (because of https://github.com/resin-io/docs/pull/345 )

lurch, 2016-10-31:
> And it's a real word as well, so capitalizing it makes it clear that Etcher is a product.

The same could be said for 'resin'.

lurch, 2016-11-02:
...and on the subject of "style guides", both the repo description and the README.md say
> Flash OS images to SD cards & USB drives, safe & easy.

whereas package.json says
> An image flasher with support for Windows, OS X and GNU/Linux.

and the website says
> Burn. Better.
Burn images to SD cards & USB drives, safe & easy.

It might be worth settling on a single "official slogan"? ;-)

WasabiFan, 2016-11-02:
Note from my pedant side: "safe & easy" should be "safely and easily" if that one is going to go into more widespread use. Alternately, it could be "It's safe and easy".

lurch, 2016-11-02:
@WasabiFan I totally agree. I'd been trying to silence my pedantism by reading the comma as a silent "it's" ;-)

jviotti, 2016-11-28:
I think this issue is covered already. Please feel free to re-open if not!

lurch, 2016-11-28:
So, if we're settling on "Etcher" rather than "etcher", does https://github.com/resin-io/docs#style-manual need to be updated?

And should the Copyright strings be saying "Resinio Ltd." instead of "resin.io" ?

jviotti, 2016-11-29:
> So, if we're settling on "Etcher" rather than "etcher", does https://github.com/resin-io/docs#style-manual need to be updated?

Re-pinging @shaunmulligan and @sonyagreen.

> And should the Copyright strings be saying "Resinio Ltd." instead of "resin.io" ?

Resinio Ltd is the company name, where "resin.io" is one product of the company. I think the copyright string (as well as the "company name" in the Windows build script) should refer to the former. @alexandrosm ?

lurch, 2016-11-29:
Re-opening issue, pending feedback from other people.

sonyagreen, 2016-11-30:
I'm hoping @alisondavis17 can weigh in on the capitalization of products (the one in question is 'Etcher' vs 'etcher').

lurch, 2016-12-04:
Now that the "slogan" seems to have been settled on as `Flash OS images to SD cards & USB drives, safely and easily.`, should that also be set as the description on the [Gitter channel](https://gitter.im/resin-io/etcher) ?

WasabiFan, 2016-12-04:
Electron's site also uses the old slogan, FYI. I don't know how if that is manual or automatic though.

lurch, 2016-12-04:
> Electron's site also uses the old slogan, FYI.

Good spot, thanks! https://github.com/electron/electron.atom.io/pull/608

jviotti, 2016-12-05:
I've updated the Gitter description!

closed 2018-10-15 by jviotti