balena-io/etcher #801: Etcher CLI broken?

lurch, 2016-10-31:
I just tried following the `docs/CLI.md` instructions (on Ubuntu 14.04), and they didn't work. So then I tried again in a totally fresh directory, and got the same error:
```bash
$ cd /tmp
$ git clone https://github.com/resin-io/etcher
$ cd etcher
$ scripts/build/linux.sh install x64
# snip long output, but it all seemed to work fine
$ bin/etcher --help
module.js:597
  return process.dlopen(module, path._makeLong(filename));
                 ^

Error: Module version mismatch. Expected 48, got 50.
    at Error (native)
    at Object.Module._extensions..node (module.js:597:18)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at /tmp/etcher/node_modules/lzma-native/index.js:14:14
    at Object.<anonymous> (/tmp/etcher/node_modules/lzma-native/index.js:626:3)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/tmp/etcher/node_modules/etcher-image-stream/lib/handlers.js:22:36)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/tmp/etcher/node_modules/etcher-image-stream/lib/index.js:26:18)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/tmp/etcher/lib/cli/writer.js:20:21)
    at Module._compile (module.js:570:32)
```

jviotti, 2016-10-31:
Hi @lurch ,

Such error means that the NodeJS version you have in your computer is not compatible with what we set in the build scripts, which kinda makes sense since the platform script build for Electron and the CLI instructions target NodeJS.

Replacing `scripts/build/linux.sh install x64` for `npm install --build-from-source` should work. I'll update the docs accordingly.

jviotti, 2016-10-31:
Oh I see, so the CLI document pointed to `CONTRIBUTING.md`, which previously had plain `npm` instructions, but then was changed to point to the platform scripts.

Actually, `npm install` without `--build-from-source` should work in the 99% of cases.

closed 2016-10-31 by jviotti