balena-io/etcher #810: Etcher CLI --help output line-wraps too early

lurch, 2016-10-31:
So, in https://github.com/resin-io/etcher/pull/802#issuecomment-257415982 I eventually managed to get `bin/etcher --help` working :smiley: 

But the output looks kinda weird:
```
$ bin/etcher --help
Usage: /usr/bin/nodejs bin/etcher <image>

Options:
  --help, -h     show help                                             [boolean]
  --version, -v  show version number                                   [boolean]
  --drive, -d    drive                                                  [string]
  --check, -c    validate write                        [boolean] [default: true]
  --robot, -r    parse-able output without interactivity               [boolean]
  --yes, -y      confirm non-interactively                             [boolean]
  --unmount, -u  unmount on success                    [boolean] [default: true]

Examples:
  /usr/bin/nodejs bin/etcher
  raspberry-pi.img
  /usr/bin/nodejs bin/etcher --no-check
  raspberry-pi.img
  /usr/bin/nodejs bin/etcher -d /dev/disk2
  ubuntu.iso
  /usr/bin/nodejs bin/etcher -d /dev/disk2
  -y rpi.img

Exit codes:
  0 - Success
  1 - General Error
  2 - Validation Error
  3 - Cancelled

If you need help, don't hesitate in contacting us at:

  GitHub: https://github.com/resin-io/etcher-cli/issues/new
  Gitter: https://gitter.im/resin-io/etcher

```

I dunno if the error is in `lib/cli/cli.js` or in the [yargs](https://www.npmjs.com/package/yargs) module, but for some reason the Examples have all been line-wrapped far too early, i.e. I'd expect them to be displayed as:
```
Examples:
  /usr/bin/nodejs bin/etcher raspberry-pi.img
  /usr/bin/nodejs bin/etcher --no-check raspberry-pi.img
  /usr/bin/nodejs bin/etcher -d /dev/disk2 ubuntu.iso
  /usr/bin/nodejs bin/etcher -d /dev/disk2 -y rpi.img
```

Also (and again this may be a limitation of the yargs module) given that I invoked the command as just `bin/etcher`, it'd probably be better if it just showed `bin/etcher` in the Examples and Usage strings, rather than displaying `/usr/bin/nodejs bin/etcher` ?

jviotti, 2016-10-31:
@lurch I checked yargs documentation and found `.wrap(null)` disables any wrapping (check the PR I just sent).

> Also (and again this may be a limitation of the yargs module) given that I invoked the command as just bin/etcher, it'd probably be better if it just showed bin/etcher in the Examples and Usage strings, rather than displaying /usr/bin/nodejs bin/etcher ?

Yeah, this is coming from `yargs`. This is how we declare the examples:

```js
.example('$0 raspberry-pi.img')
.example('$0 --no-check raspberry-pi.img')
.example('$0 -d /dev/disk2 ubuntu.iso')
.example('$0 -d /dev/disk2 -y rpi.img')
```

`$0` is replaced automatically by yargs. Maybe there is a way we can somehow tweak it, or compute the string ourselves. If we find a better solution, we can always send a patch upstream to yargs :)

BTW, as a nice trick: if you want to run the CLI but don't want to re-install the dependencies that have been installed by the build scripts, you can do the following:

```sh
ELECTRON_RUN_AS_NODE=1 ./node_modules/.bin/electron bin/etcher
```

closed 2016-10-31 by jviotti

jviotti, 2016-11-01:
Yeah, `yargs` is a bit weird, but has a lot of functionality that simplifies things for us. Is also quite minimal, so it serves our purposes well. I guess the way to go is to PR functionality to `yargs` itself to improve it on this side.