balena-io/etcher #818: Can't globally install on linux

agherzan, 2016-11-02:
- 1.0.0-beta.16
- Linux carbonA 4.8.2-1-ARCH #1 SMP PREEMPT Mon Oct 17 08:11:46 CEST 2016 x86_64 GNU/Linux
- NO

Trying to global install fails with:

```
npm ERR! Linux 4.8.2-1-ARCH
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "-g"
npm ERR! node v6.9.1
npm ERR! npm  v3.10.9
npm ERR! code EBADPLATFORM

npm ERR! notsup Unsupported platform for appdmg@0.3.10: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm ERR! notsup Valid OS:    darwin
npm ERR! notsup Valid Arch:  any
npm ERR! notsup Actual OS:   linux
npm ERR! notsup Actual Arch: x64

npm ERR! Please include the following file with any support request:
npm ERR!     /home/andrei/work/resin/etcher/npm-debug.log
npm ERR! code 1
```

jviotti, 2016-11-02:
Hey there,

What global package are you trying to install? We're not currently
publishing anything to NPM. In case you're trying `etcher-cli`, that is
very deprecated package that is not being maintained anymore.

Check https://github.com/resin-io/etcher/blob/master/docs/CLI.md for
offical instructions on how to get the Etcher CLI up and running.

agherzan, 2016-11-02:
Got it. So we don't publish the package anymore. I will close the bug in this case as this is invalid now.

closed 2016-11-02 by agherzan