balena-io/etcher #827: "wrong" LICENSE file in deb packages?

lurch, 2016-11-03:
As I originally mentioned in https://github.com/resin-io/etcher/issues/632#issuecomment-257935061 , if you install the `etcher-electron` package from the resin.io bintray [repository](https://bintray.com/resin-io/debian/etcher), then `/usr/share/etcher-electron/LICENSE` says:
```
Copyright (c) 2016 GitHub Inc.

Permission is hereby granted, free of charge, to any person obtaining
a copy of this software and associated documentation files (the
"Software"), to deal in the Software without restriction, including
without limitation the rights to use, copy, modify, merge, publish,
distribute, sublicense, and/or sell copies of the Software, and to
permit persons to whom the Software is furnished to do so, subject to
the following conditions:

The above copyright notice and this permission notice shall be
included in all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND,
EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF
MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND
NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE
LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION
OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION
WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
```

This makes it look like etcher was created by GitHub, and licensed under MIT, which probably isn't what we want? IMHO that file ought to be https://github.com/resin-io/etcher/blob/master/LICENSE

ping @dlech

dlech, 2016-11-03:
FWIW, this file is included in all of the packages for all OSes, not just debian.

And this license is still applicable as well and we would be in violation if we did not include it. It should probably just be renamed to `LICENSE.electron` or something like that. And technically, there is probably a license for every one of the many hundreds of dependencies that should be included as well.

I know nothing about how the electron build system, but if there is a way to include arbitrary files in the final package, then we could include the main Etcher license.

jviotti, 2016-11-03:
There are `Etcher-$VERSION-$OS-$ARCH` directories at `etcher-release` as a consequence of the "package" command in every operating system, so this fix can be applied during the "package" phase.

lurch, 2016-11-03:
> There are `Etcher-$VERSION-$OS-$ARCH` directories at etcher-release as a consequence of the "package" command in every operating system

That's not necessarily true - for Linux it's `Etcher-$OS-$ARCH` and it's only when creating the zipfile that the `$VERSION` gets included.

jviotti, 2016-11-28:
This issue was fixed as part of ditching `electron-packager` in https://github.com/resin-io/etcher/pull/897.

closed 2016-11-28 by jviotti