balena-io/etcher #828: Problem with USB drive appearing as "system" drive

lurch, 2016-11-03:
- **Etcher version:** 1.0.0-beta.16
- **Operating system and architecture:** OSX
- **Do you see any meaningful error information on DevTools?** Unknown

Despite asking him to do so, @PrintTrader never opened a GitHub issue after I was trying to help him (via Gitter) diagnose why Etcher wasn't working properly for him.
Rather than repeating everything here, I'll just post a link to the chat log, so that this issue doesn't get "lost". But I appreciate it may be difficult to make any progress without further debugging info.

https://gitter.im/resin-io/etcher?at=5819d9c8806316005deb8368  (conversation stops at Nov 02 14:51)

lurch, 2016-11-03:
https://github.com/resin-io-modules/drivelist/issues/97 also seems to be showing a USB drive as a "system" drive.

jviotti, 2016-11-03:
@lurch Thanks a lot! You're right that we can't do much without any more information, but lets keep the issue open for a while in case the user appears on GitHub again, and you can get his `diskutil info /dev/diskN` command output.

lurch, 2016-11-03:
Also had [this](https://gitter.im/resin-io/etcher?at=581b731645c9e3eb43f3c2f9) similar sounding report on Gitter today (on Windows 10 this time) but there wasn't even a second follow-up message.

jviotti, 2016-11-18:
I'm closing this, since there is no way to move forward without any further information. Lets re-open if the users provide more insight!

kevincraig, 2016-12-04:
@jviotti 
Just as an FYI I had this issue this morning.  With a SanDisk 32gb USB flash drive.  I had just erased it via Disk Utilities on Mac OS X 10.11.6 below is what darwin output for me.  You can see that this drive as well shows as a System Drive.  

When I erased it I used the format "OS X Extended (Journaled)" and Scheme GUID Partition Map

device: /dev/disk4
description: USB Flash Drive
size: 32015679488
mountpoint: /Volumes/UNTITLED
name: /dev/disk4
raw: /dev/rdisk4
protected: False
system: True

Hope this helps if not reach out to me I'll be more than happy to help diagnose the issue.

lurch, 2016-12-04:
Thanks @kevincraig , can you also paste the output of the commands listed [here](https://github.com/resin-io/etcher/issues/649#issuecomment-241728425) please?

kevincraig, 2016-12-04:
Hope this helps.  Let me know if there is anything else I can help you with.  Totally understand running down bugs/issues.  

<pre><code>

/dev/disk4 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *32.0 GB    disk4

</code></pre>

<pre><code>
 Device Identifier:        disk4
   Device Node:              /dev/disk4
   Whole:                    Yes
   Part of Whole:            disk4
   Device / Media Name:      USB Flash Drive

   Volume Name:              Not applicable (no file system)

   Mounted:                  Not applicable (no file system)

   File System:              None

   Content (IOContent):      GUID_partition_scheme
   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 USB
   SMART Status:             Not Supported

   Total Size:               32.0 GB (32015679488 Bytes) (exactly 62530624 512-Byte-Units)
   Volume Free Space:        Not applicable (no file system)
   Device Block Size:        512 Bytes

   Read-Only Media:          No
   Read-Only Volume:         Not applicable (no file system)

   Device Location:          External
   Removable Media:          No

   Virtual:                  No
   OS 9 Drivers:             No
   Low Level Format:         Not supported

</code></pre>

lurch, 2016-12-04:
So, the same situation as in #649 - `diskutil` is reporting
```
Removable Media:          No
```
which makes Etcher think it's a "system" drive :-/

I'll leave this issue open in case @jviotti has any other ideas or debugging steps @kevincraig can try and perform...

jviotti, 2016-12-05:
There are a couple of things we *ciykd* do here (a combination of them is possible as well). We can mark a drive a `system: false` in the following cases:

- The drive has no file system (like the drive in question). If it doesn't have a file-system, then we could assume we can write to it without destroying any data, however it could be that the drive is formatted using a file system that OS X doesn't recognize. In any case, showing a hard drive with no file system as `system: false` would be still really awkward, so we could combine with something like "External".

- Mark a device as `system: false` if `Device Location` is `External`, however this means that external hard drive will be now treated as removable drives.

In any case, none of the options mentioned above are accurate enough to be included in the project IMHO. There might be another heuristic we could take in this case, but for the time being, I suggest closing this. The OS for some reason thinks the drive is not removable, and trusting the OS is the best we can do at the moment.

closed 2016-12-05 by jviotti

lurch, 2016-12-05:
> The OS for some reason thinks the drive is not removable, and trusting the OS is the best we can do at the moment.

Yup. And this will obviously become far less of an issue when we change Etcher to always display system-drives #888

thenetimp, 2017-08-13:
any word on this?  I have a 80GB SATA drive connected via USB that Etch doesn't recognize.

I get the problem of not wanting to write to system drives, but shouldn't you give the end user the ability to make that decision instead of deciding for them?  At least make it an option that can be disabled?

thenetimp, 2017-08-13:
Device Identifier:        disk2
   Device Node:              /dev/disk2
   Whole:                    Yes
   Part of Whole:            disk2
   Device / Media Name:      Jmicron Media

   Volume Name:              Not applicable (no file system)
   Mounted:                  Not applicable (no file system)
   File System:              None

   Content (IOContent):      FDisk_partition_scheme
   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 USB
   SMART Status:             Not Supported

   Disk Size:                80.0 GB (80026361856 Bytes) (exactly 156301488 512-Byte-Units)
   Device Block Size:        512 Bytes

   Read-Only Media:          No
   Read-Only Volume:         Not applicable (no file system)

   Device Location:          External
   Removable Media:          Fixed

   Virtual:                  No
   OS 9 Drivers:             No
   Low Level Format:         Not supported

lurch, 2017-08-14:
> At least make it an option that can be disabled?

Yup, that's exactly what enabling the "Unsafe mode" in Etcher's settings does :grinning:

thenetimp, 2017-08-14:
Ah, see that wasn't clear looking at the controls, and I am lazy and didn't go searching for docs. Thanks for the update I'll look into it later.

lurch, 2017-08-14:
Yup, see #1432

d9media, 2023-04-02:
Apologies for hi-jacking this thread but I couldn't access the mentioned chat log in the original post. I used etcher to write Clonezilla to a usb and the process got interrupted when the pc went down and showed a black screen just saying ”no bootable disk found”. Yes, Etcher wrote the image to my system disk. I desperately searched for similar threads but couldn't find any and this issue sounded somewhat related. 
Are you guys aware of any issues like that in the past? 

Is it even possible to select your main system disk as a target? (I certainly don't remember picking it!) 
I'm not even gonna ask for a possible path of recovery, I pretty much accepted that this will be impossible to fix. I tried cloning the drive with Clonezilla and it immediately aborts with the message "