balena-io/etcher #829: Package Etcher for Windows Store

WasabiFan, 2016-11-03:
With Project Centennial, Microsoft introduced the ability to package standard win32 apps as AppX packages which could be published to the Windows Store. Packages produced in this way go through the standard Store ingestion process and contain raw win32 executable bits, however at run-time the system launches the app in a sandbox with certain virtualized resources (e.g. the registry). Tools have been developed expressly for the purpose of packaging Electron apps with this new technology, so theoretically this opens the door for Etcher to be published to the Store.

I don't know much about the restrictions put in place by their UWP sandbox, but I suspect that there's much more than just the registry and certain file system locations as their documentation mentions. I'd imagine that it wouldn't react well to apps attempting to run VBScript or access drive hardware as Etcher does, but it's always a possibility. I think it would be interesting to pursue packaging Etcher for the Store, because with that comes the ability for one-click installation and automatic updates (not to mention the inherent trust which comes from downloading an app from the Store).

Here are some resources I found on the topic (new things may have been published since I found these):
- https://github.com/felixrieseberg/electron-windows-store
- http://electron.atom.io/docs/tutorial/windows-store-guide/
- https://msdn.microsoft.com/windows/uwp/porting/desktop-to-uwp-behind-the-scenes

jviotti, 2016-11-03:
Sounds interesting! I'll give it a go as soon as I can allocate some time for it!

WasabiFan, 2017-01-21:
Desktop apps are now being allowed more broadly into the Windows Store, and I see no reason that Etcher wouldn't be technically functional if packaged with their new tools. Apps ranging from Slack (which is really just their website in an Electron frame) to lower-down system utilities have been demonstrated to work, and they show up with a new privilege level which is essentially "do whatever it wants." Even the desktop app which allows you to package desktop apps is in the Store, and it's a CLI :laughing: You can see the docs page here: https://msdn.microsoft.com/en-us/windows/uwp/porting/desktop-to-uwp-run-desktop-app-converter. Essentially, their new tools run your installer in a sandbox and package the file system diff as an app; I am not entirely sure of the specifics of its privileges at runtime, though.

It would be really cool if you were able to try it out and see if it works! If this turned out to work properly, the optimal Windows installation instructions could be streamlined to two clicks (one on the Etcher website and once in the Store for confirmation). No rush though.

alexandrosm, 2017-01-22:
> allows you to package desktop apps is in the Store, and it's a CLI 😆 You
> us/windows/uwp/porting/desktop-to-uwp-run-desktop-app-converter.

for context, what is the installation experience for windows store apps?

WasabiFan, 2017-01-22:
1. The user arrives at the page for the program in question in the Store app. The Store app responds to custom protocol links, so you can link from a webpage directly to the Store.
2. The user clicks the "get" button.
3. Either the app begins installing or the user is asked to confirm once before it installs; I believe this depends on the version of the OS.
4. The user is notified when the app finishes installing.

The app installation packages are all hosted by Microsoft's CDNs; to get an app in the Store, you submit it for approval and they test it to confirm that it complies with their policies before pushing it for public use.

If you have a Windows 10 PC available, I'd recommend trying it out so that you can get a feel for it :wink:

jviotti, 2017-01-23:
Great stuff, we should definitely look into this soon. I've used the Windows Store a few times, and I agree that the installation experience is much better than downloading an exe and going through a wizard!

luksireiku, 2017-10-03:
Any progress with this? an UWP app that autoupdates it's better than having to download and install everytime that a new update gets released.

TheZoker, 2020-05-06:
Any news on this?

hansmbakker, 2022-01-24:
https://winget.run/pkg/Balena/Etcher is also available, but still having a Store entry would be nice!

mi-hol, 2022-08-19:
winget package seems to have issues with upgrades

I've installed version 1.5.84 in Jan 2020 and when I run the command `winget upgrade --all --include-unknown`

```
Name                                   Id                                Version          Available      Source
---------------------------------------------------------------------------------------------------------------
... (removed irrelevant info)
balenaEtcher 1.5.84                    Balena.Etcher                     1.5.84           1.7.9          winget
7 upgrades available.

... (removed irrelevant info)
(5/5) Found balenaEtcher [Balena.Etcher] Version 1.7.9
This application is licensed to you by its owner.
Microsoft is not responsible for, nor does it grant any licenses to, third-party packages.
Downloading https://github.com/balena-io/etcher/releases/download/v1.7.9/balenaEtcher-Setup-1.7.9.exe
  ██████████████████████████████   138 MB /  138 MB
Successfully verified installer hash
Starting package install...
Successfully installed
```

it installed version  1.7.9 but an entry for 1.5.84 remained in Apps.

[image]

Selecting "Deinstall" 1.5.84 actually removes 1.7.9 and 1.5.84 remains as orphaned entry that shows below error after selecting "Deinstall"

[image]