balena-io/etcher #830: Drive-filtering suggestion

lurch, 2016-11-03:
After https://github.com/resin-io-modules/drivelist/issues/98 has been implemented (i.e. we can get info about _all_ the mountpoints which are used by each physical drive), we could filter out the drive on which the image we're going to be flashing is stored on.

Currently, this scenario is possible (which would be prevented if this issue is implemented) :
 1. User has a USB Flash drive (e.g. `/dev/sdb`) and a USB SD-reader (e.g. `/dev/sdc`) connected to their computer
 2. User is running out of space on their computer's HDD, so they download e.g. `2016-09-23-raspbian-jessie-lite.zip` to their USB Flash drive
 3. User launches Etcher, and selects the `2016-09-23-raspbian-jessie-lite.zip` file they've stored on their USB Flash drive
 4. User accidentally selects `/dev/sdb` as the drive they want to flash to
 5. User clicks `Flash!`
 6. ???? (I didn't actually try this myself to see what happens)

If this issue was fixed, `/dev/sdb` would automatically be filtered out (until the user selects an image _not_ stored on their USB Flash drive), and Etcher would auto-select `/dev/sdc`.

jviotti, 2016-11-03:
Very good observation, we should definitely be doing this, so I'll mark it as a bug.

jviotti, 2016-11-09:
Should we omit the drive altogether, or display a warning label somehow describing that the drive is the place where the image resides?

This is what we do for locked SD Cards for example (although this is an old screenshot the labels are being shown below the drive information at the moment):

[image]

@konmouz ?

WasabiFan, 2016-11-09:
Side-note: It would be great if labels like this "locked" one explained _why_ the device is locked. I could easily imagine a user looking at this and thinking that _Etcher_ locked it as opposed to the device being locked by the OS/hardware. Is there a tooltip already there that serves this purpose?

jviotti, 2016-11-09:
Very good point! Something for @konmouz and @taahirisaacs to help us with.

lurch, 2016-11-09:
I noticed this evening that an 'unselectable' drive (in my specific case I was experimenting with an image bigger than the drive) in that dialog still displays a hand-cursor when you mouse-hover over it, which it probably shouldn't.

jviotti, 2016-11-11:
@lurch Good catch, we should correctly set the `cursor` CSS property there. Do you mind opening an issue for it?

lurch, 2016-11-11:
> @lurch Good catch, we should correctly set the cursor CSS property there. Do you mind opening an issue for it?

Done, #865

alexandrosm, 2017-01-25:
@jviotti should we close this?

jviotti, 2017-01-26:
I think this feature is still valid, and haven't been done yet.

konmouz, 2017-01-27:
@jviotti we can follow #993 here or just a tooltip. Please, also check [zeplin](https://zpl.io/Z2LGn4) for label reference.

jviotti, 2017-01-27:
@konmouz Not sure I'm seeing the Zeplin screen you're referring to. How would a tooltip/popover work here? Maybe we should disable such drive with a nice badge that somehow conveys to the user that this is the drive that contains the image instead?

lurch, 2017-01-27:
> Maybe we should disable such drive with a nice badge that somehow conveys to the user that this is the drive that contains the image instead?

That's what I was thinking, yes.

jviotti, 2017-01-27:
That is an easy fix. The hard part is to figure out what to put in the badge. I can't think of anything short while descriptive. "IMAGE IS IN DRIVE"?

lurch, 2017-01-27:
"SOURCE DRIVE" ?

konmouz, 2017-01-31:
@jviotti I am referring to drive labels in general (e.g zeplin page: Step 2/select drive/modal/inactive). Also, labels should be below title, not above - if possible.
@lurch "SOURCE DRIVE" sounds great. We can use just a normal black tooltip for details.

jviotti, 2017-01-31:
@konmouz What do you think of https://github.com/resin-io/etcher/pull/1066?

closed 2017-02-06 by jviotti