balena-io/etcher #833: There's no "all files" filter available at image selection

imrehg, 2016-11-04:
- **Etcher version:** 1.0.0-beta.16
- **Operating system and architecture:** ArchLinux x64, XFCE

When trying to open an image to burn, there are no something like "show all files" option in the dropdown, only "OS Images" value. This makes it difficult to select files that are actually images, but not named following the expected filters. E.g. the `.resin-sdcard` output of the resin.io image maker process, I had to rename the file to be able to burn it, which should not be required.

jviotti, 2016-11-07:
@imrehg Thanks for reporting this!

We pass to Electron a list of extensions we support, which is then used by the framework to display the dialog. I'll check if there is a way to specify to Electron that we still want to allow the user to select anything they want, but it might be that we will need to reach out to the Electron team for this.

In any case, even if we allow the user to throw any format at it, Etcher business logic will reject an image as invalid if it doesn't fall inside our list of supported formats, so we also need to somehow make the logic more flexible, while still strict enough to prevent the user from doing something that doesn't make sense. For example: a user might want to select a `.7z` file in Etcher. Since the file will get accepted, the user will assume decompression will take place, while that won't be the case, since that format is not supported by Etcher at the moment.

Obviously an alternative would be to whitelist the `.resin-sdcard` extension, which is trivial.

What do you think?

dlech, 2016-11-07:
Maybe you could change the Etcher logic so that if the user changes the filter, it handles the file differently. You could call the no filter/all files option "raw images" or something like that and just write the file verbatim with no special handling when the raw option is selected.

lurch, 2016-11-08:
Maybe there could be an advanced option in the settings, which if ticked (it'll be off by default) will make Etcher behave as @dlech describes above? Also means that there'll be no need to change the extension-filtering logic in Electron, as if the option is ticked it'll just display `*.*`.

jviotti, 2016-11-08:
Something like this seems to do the trick:

```diff
diff --git a/lib/gui/os/dialog/services/dialog.js b/lib/gui/os/dialog/services/dialog.js
index 1b2d0f4..afeb772 100644
--- a/lib/gui/os/dialog/services/dialog.js
+++ b/lib/gui/os/dialog/services/dialog.js
@@ -58,6 +58,10 @@ module.exports = function($q, SupportedFormatsModel) {
           {
             name: 'OS Images',
             extensions: SupportedFormatsModel.getAllExtensions()
+          },
+          {
+            name: 'All Files',
+            extensions: [ '*' ]
           }
         ]
       }, (files) => {
```

WasabiFan, 2016-11-08:
Honestly, I think that adding an "all files" option would confuse users more than it helps. If Etcher relies on the extension to determine the file type (and implicationally how to handle it), adding this option will give the impression of supporting any file type, even though in reality it doesn't. The contract between an application and its user is that it will allow the user to select files which are appropriate for use, and prevent selection off ones which are not. If a user attempts to select one of those mac-specific file types which aren't supported, for example, that would end up with a successful flash only to not be able to boot their drive (because Etcher silently decided to dump the file contents directly onto the disk). I believe the more appropriate action would be to change the file that you distribute to be a `.etch` file or add the current extension to the allow list.

Shou, 2016-11-21:
We could label the selection mode as 'All files (unsupported)', or maybe this is something 'unsafe mode' can toggle? Though maybe grouping too many features under 'unsafe mode' is getting a bit too implicit to the user as the exact behaviour of 'unsafe mode' would have to be explained further.

jviotti, 2016-11-21:
> Honestly, I think that adding an "all files" option would confuse users more than it helps. If Etcher relies on the extension to determine the file type (and implicationally how to handle it), [...]

I completely agree with this. Etcher looks at the MIME type rather than the extension, and will throw "Invalid image" if the MIME type is not recognized, but the problem is that many unrecognized image formats will be interpreted as `application/octet-stream`, causing Etcher to perform usual byte-by-byte flashing without any data transformation (e.g: decompression), so while really incorrect formats will be caught early (e.g: someone trying to flash a `.jpeg` "image"), some will not get caught at all, and the user will have a really hard time trying to figure out why.

> We could label the selection mode as 'All files (unsupported)', or maybe this is something 'unsafe mode' can toggle? Though maybe grouping too many features under 'unsafe mode' is getting a bit too implicit to the user as the exact behaviour would have to be explained further.

Definitely not something to be included inside "unsafe mode". The "All files (unsupported)" filter looks good at first sight, however remember that if the user tries to select any file, he will be able to select both supported *and* unsupported files, in which case the "unsupported" label stop making much sense.

It also gives the impression that the "All files" dialog functionality is broken or incomplete.

jviotti, 2016-11-21:
I vote to close this issue. The main reason for this improvement is the usage of a `.resin-sdcard` extension, which is definitely non-standard, and in theory, is really an `.img` file. What do you think?

alexandrosm, 2016-11-21:
The only thing we could do here is have a "disable filetype filtering"
option in the settings, that will cause etcher to literally take any file
and write it to the card, with the user's responsibility. whether we want
that or not, is a different matter.

jviotti, 2016-11-23:
I'll close this down since the use cases for this feature are very few, it can get really confusing, and the user can always just rename the extension or something (so the user is not blocked).

closed 2016-11-23 by jviotti