balena-io/etcher #834: EIO: i/o error, write on beta 16

Ntemis, 2016-11-06:
Hello,
Am getting a lot of
Error: EIO: i/o error, write
    at Error (native)-
If i delete partitions on sdcard manually and try to write again sometimes is a hit or miss but error goes way.
 
**Etcher version:** beta16
- **Operating system and architecture:** ubuntu 16.04
- **Do you see any meaningful error information on DevTools?**

lurch, 2016-11-06:
Can you provide more details about your SD card?
Is it possible it's a fake, or is faulty? Or if the SD card isn't faulty, perhaps your SD card-reader is faulty?

Have a look at http://oss.digirati.com.br/f3/ - if that reports errors with your SD card&reader combo then I don't think there's much that Etcher can do.
If you have multiple SD cards, or another SD reader, it's always worth swapping things about to see if that improves things :)

jviotti, 2016-11-07:
Yeah, it sounds like a faulty SDCard, or SDCard reader. Lets do some further investigation if this keeps failing after retrying with a different SDCard or reader.

As a side note, maybe we should intercept EIO error and output a more friendly message?

jviotti, 2017-01-04:
I believe this issue has been fixed with https://github.com/resin-io-modules/etcher-image-write/pull/73

hkobir, 2019-12-18:
in my case error has been solved.
note: Be sure that img file has been extract with 7zip.