balena-io/etcher #844: etcher stuck on the "Starting..." state when trying to burn the same image a second time

rgl, 2016-11-07:
- **Etcher version: 1.0.0-beta.16**
- **Operating system and architecture: Windows 10 amd64**
- **Do you see any meaningful error information on DevTools?**

```
Mon Nov 07 2016 18:18:35 GMT+0000 (GMT Standard Time) Restart ({"preserveImage":true}) tracker.js:10
Mon Nov 07 2016 18:18:39 GMT+0000 (GMT Standard Time) Flash ({"image":"C:\\Users\\rui.g.lopes\\Projects\\rgl\\debian-live-builder-vagrant\\live-image-amd64.hybrid.iso","device":"\\\\.\\PHYSICALDRIVE2"})

C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\node-ipc\dao\socketServer.js:287 
Uncaught TypeError: this.log is not a function
  (anonymous function) @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\node-ipc\dao\socketServer.js:287
  emitOne @ events.js:96
  emit @ events.js:188
  emitErrorNT @ net.js:1272
  _combinedTickCallback @ internal/process/next_tick.js:74
  _tickCallback @ internal/process/next_tick.js:98
```

rgl, 2016-11-07:
Oh, its not a matter of using the same image. With different image the same thing happens. So this always happens when trying to burn another image without quitting etcher.

lurch, 2016-11-08:
I believe this was fixed in #816 (but isn't yet available in a pre-built Etcher image).

jviotti, 2016-11-08:
Hi @rgl ,

Thanks for reporting! As @lurch mentioned, this issue has been indeed fixed in master, and it will land the next release! Sorry for the inconveniences!

closed 2016-11-08 by jviotti

lurch, 2016-11-18:
I found an even faster way to reproduce this last night, without having to wait for a drive to finish flashing first... ;)
Run Etcher-portable on a Windows account with limited privileges. Click the Flash button, but then click No in the elevation prompt. Click the Flash button again and you immediately get the button stuck on "Starting..." and `this.log is not a function` in devtools console.

jviotti, 2016-11-18:
@lurch So is it still happening on `master`? Even after https://github.com/resin-io/etcher/pull/816?

lurch, 2016-11-19:
> So is it still happening on `master`?

Dunno, I'm not yet set up to build Etcher from source on Windows (ran out of space on my Windows laptop).

Monis1, 2017-07-22:
I ran into the same issue in the first attempt, after selecting the .iso file and usb drive , I clicked on the Flash button, it got stuck on starting...shows the progress in the second attempt...

lurch, 2017-07-22:
@Monis1 Please open a new issue describing your problem in more detail. Thanks.

mastercake10, 2017-10-21:
Same issue here.

cebess, 2017-12-09:
Down loaded the latest build this morning and had the same issue dec-19-2017

lurch, 2017-12-11:
@cebess This issue has been closed, and is actually slightly different to the issue you're experiencing.
Please open a new issue describing your problem in as much detail as possible, thank you.