balena-io/etcher #845: Drive left read only

cardonator, 2016-11-07:
- **Etcher version:** 1.0.0-beta16
- **Operating system and architecture:** Ubuntu 16.04 64-bit
- **Do you see any meaningful error information on DevTools?** No

I was using Etcher to make a FreeDOS bootable USB drive which worked great except that it created the only partition in an indeterminate file system format and whenever I would try to write files to the drive I would get an error back because the system was read only (write protected).

I tried to use hdparm to set readonly to off but it had no effect. I was unable to fix the problem to copy a file to the drive. I ended up going back to Unetbootin (sadly).

lurch, 2016-11-08:
Can you provide a link to the FreeDOS image you tried flashing to your USB drive? Apart from being read-only, did it actually boot?

jviotti, 2016-11-08:
Etcher simply copies the image content byte by byte, so the read-only setting probably comes from the image itself. UNetbootin takes a very different approach, since it extracts the files from the image, partitions the drive itself, and copies things over, so that's probably the reason why you're not hitting it in UNetbootin.

I'll close this issue under the assumption that the actual image is causing the final drive to be read-only, but feel free to re-open if you find this is not the case (you could confirm with `dd`, for example, which takes a very similar writing strategy to us).

closed 2016-11-08 by jviotti

cardonator, 2016-11-08:
I used FreeDOS 1.0, 1.1, and 1.2 and all exhibited the same issue. I'm not so concerned that the image writes as read only. It was/is more annoying that I can't seem to unset the read only flag on the drive afterwards.

I also tried doing this with multiple other ISOs. They all seemed to do the same thing. However, just to compare I did try writing them all with dd and noticed the same problems going on so I suspect that it is not anything exclusively related to Etcher after all.

That being said, is it possible for Etcher to unset the readonly flag during the write process somehow? It might be a nice option.

jviotti, 2016-11-09:
> That being said, is it possible for Etcher to unset the readonly flag during the write process somehow? It might be a nice option.

I guess we could, in theory (at least in some operating systems), however that could lead to some weird results (the readonly flag might be there for some reason).

That being said, I guess we can investigate some ways to unset this and document the solution in Etcher's user documentation.