balena-io/etcher #846: Add the UAC Shield to the flash button in Windows.

jviotti, 2016-11-08:
According to Windows guidelines (https://msdn.microsoft.com/en-us/library/windows/desktop/dn742497(v=vs.85).aspx), buttons that trigger an elevation mechanism should contain the well-known UAC shield icon.

Maybe displaying the UAC shield icon as it is would not be very pretty, but we could build a similar icon that better adapts to Etcher style guide, while still makes it clear to user that the action requires elevation.

[image]

1j01, 2016-11-17:
There's the [Windows 10 version](https://www.google.com/search?q=uac+icon+windows+10&tbm=isch#imgrc=uh_DblIOpxaNaM%3A) of the icon, which might fit in better, or at least work as a better base for creating an icon. Here's what it looks like at 16px (without any strokes removed): [image]

WasabiFan, 2016-11-17:
I agree with @1j01 -- you should make sure to use the win10 version.

alexandrosm, 2016-11-18:
now here's a question -- there's an architecture whereby Etcher would only
ask for elevation during the first write, and keep it for subsequent ones.
Would the subsequent writes also have the shield on the button?

WasabiFan, 2016-11-18:
I think the idea is (or at least should be) to follow Microsoft's UX guidelines. Some excerpts from [here](https://msdn.microsoft.com/en-us/library/windows/desktop/dn742497(v=vs.85).aspx):

>  Tasks that require immediate elevation when UAC is fully enabled have entry points marked with the UAC shield. In this case, users should expect to see an Elevation UI immediately after clicking such commands—and they should be extra cautious when they see Elevation UI from tasks that don't have a shield. 

---

> Note that tasks don't "remember" elevated states. For example, if the user navigates back and forth over an elevation entry point in a wizard, the user must elevate each time.

---

> Display controls with the UAC shield to indicate that the task requires immediate elevation when UAC is fully enabled, even if UAC isn't currently fully enabled. ... Proper use of the UAC shield helps users predict when elevation is required.

---

> Once elevated, stay elevated until elevated privileges are no longer necessary. Users shouldn't have to elevate multiple times to perform a single task.

alexandrosm, 2016-11-18:
If I read this correctly, we should have the shield only if there will be
an elevation prompt right after clicking.

WasabiFan, 2016-11-18:
That's my impression too. However, it does make you consider the fact that this "elevate once" model could be unpredictable for the user: It asks you for administrator privileges once, and then without you knowing holds on to that privilege. It might feel safer if there were a checkbox that allowed you to "stay elevated" that would make the behavior clearer, but at the same time that could be more confusing than helpful.

1j01, 2016-11-18:
I actually think the button should keep the icon, considering you're encouraged to display the icon even when UAC isn't fully enabled (in which case it wouldn't display the prompt):

> Display controls with the UAC shield to indicate that the task requires immediate elevation when UAC is fully enabled, even if UAC isn't currently fully enabled

It's already the case that buttons with the shield icon don't always show the prompt, i.e. in Window's Settings when you're an administrator.

jviotti, 2016-11-18:
> Note that tasks don't "remember" elevated states. For example, if the user navigates back and forth over an elevation entry point in a wizard, the user must elevate each time.

I think this means that according to Windows' user guide, we shouldn't be remembering elevation. Thoughts?

lurch, 2016-11-18:
> Thoughts?

Yeah, `Note that tasks don't "remember" elevated states.` and `Once elevated, stay elevated until elevated privileges are no longer necessary.` can be seen as somewhat contradictory :confused: 
Hard to tell what the correct behaviour is - do @konmouz and @taahirisaacs have any thoughts?

(Just to clarify, Etcher currently uses the "elevate each time the user clicks on the 'Flash!' button" model, but I wondered in #851 if that should be changed. Unless there's very good reasons not to, IMHO Etcher should use the same behaviour on all three platforms, for a consistent UX.)

WasabiFan, 2016-11-18:
I think that the "keep privileged process running" approach is fine, but it should be predictable for the user. I could see a checkbox allowing you to choose the behavior or a notice telling you what will happen; or maybe you break it into two steps: you click "Elevate" and then the button changes to "flash" (or maybe there'd be two buttons). Each option would make it clear to the user that Etcher will maintain its right to mess with their system after it finishes its primary action.

lurch, 2016-11-19:
@WasabiFan If we give the user too many options (extra checkboxes or multiple buttons), we run the risk of losing the 'simplicity' of Etcher that people appreciate so much :-/

It's worth bearing in mind that the "elevation dialog" (i.e. where it asks you for your password if you're running Etcher as a non-privileged user) is entirely OS-controlled, so we don't have any options to customise it, so any 'extra options' would need to be squeezed into the main Etcher UI.

WasabiFan, 2016-11-19:
> If we give the user too many options (extra checkboxes or multiple buttons), we run the risk of losing the 'simplicity' of Etcher that people appreciate so much :-/

I was presenting options to solve the problem at hand, not necessarily net _better_ options 😆 I think that, optimally, there would be something to make the behavior clear to the user (it asking for permissions the first time but not the second is confusing). How about this: what if Etcher changed its window title to "Etcher (Administrator)" once you had elevated? Of course, this text would change depending on the platform. I think this would let the user know what was going on while still being unobtrusive.

1j01, 2016-11-19:
I don't think it's that confusing; if a user even notices that they weren't asked to elevate the second time, they can think, "well, that's probably because I already elevated." The way to think of it is the shield means an action requires elevated priviledges, not that it *will ask you* for priviledges.
So I think we can really just have the icon on the button.

WasabiFan, 2016-11-19:
The fact that you need to explain "the way to think of it" means that there's something we need to explain.

> ...an action requires elevated priviledges, not that it will ask you for priviledges.

I'll refer you to the Microsoft docs, as quoted above.

> Display controls with the UAC shield to indicate that the task requires immediate elevation when UAC is fully enabled, even if UAC isn't currently fully enabled. ... Proper use of the UAC shield helps users predict when elevation is required.

---

UX is a game of alleviating as many potential sources of confusion or difficulty for the end user as possible. Making behaviors clear so that the user can predict what your app will do is a big part of that.

alexandrosm, 2016-11-19:
here's another idea entirely:

Etcher elevates at the point of writing because there were issues with
elevating the main process at startup.

That, however, is a technical limitation, and in terms of interaction, if
we were going to retain elevation, we should do it app-wide, not in a way
that can be imputed as sneaky (claim to want it for a specific write, then
keep it)

So what if Etcher started a persistent, elevated sub-process right at
startup, as the first thing the UI did. In that case we would need no
buttons with shields, etc.

In fact, if we do this right, we could use the elevation dialog and the
associated user interaction to "hide" some of Etcher's startup lag.

thoughts?

jviotti, 2016-11-19:
> So what if Etcher started a persistent, elevated sub-process right at
> startup, as the first thing the UI did. In that case we would need no
> buttons with shields, etc.
>
> In fact, if we do this right, [...]

Only displaying the dialog when its really necessary (right before a
flash) provides a better experience IMHO. Having an extra step at
startup can be very annoying (I remember some users complaining about it
on Etcher in the early days), and its not what the Microsoft guidelines
recommend (they recommend the shield, so you can ask for elevation only
when really needed).

***

After reading the whole thread, preserving or not preserving elevation
seems like a tricky decision where most of the alternatives are
confusing in one way or another. As far as I can see, if we want to go
for "simplicy" (from the user's point of view), then the simplest
approach is to just ask for elevation every time is needed (as we do it
now). The current behaviour is crystal clear and predictable, and its
only draw back is an extra button click.

I think remembering elevation is a symptom, and not the root cause of
the problem. The main use case for remembering elevation is for users
who want to flash an image many times in a row, which is a problem that
will be better addressed by the "multi-write" feature we're currently
working on.

What do you think?

lurch, 2016-11-19:
> The main use case for remembering elevation is for users
who want to flash an image many times in a row, which is a problem that
will be better addressed by the "multi-write" feature we're currently
working on.

Yeah, I suspect that the proportion of Etcher users who'll be writing multiple images in a row is quite small, so perhaps for the sake of simplicity #851 should be closed?

jviotti, 2016-11-19:
Yeah I agree, let's close that one and solve the underlying problem with multi writes support.

closed 2018-10-15 by jviotti