balena-io/etcher #848: Increase image drag and drop area

jviotti, 2016-11-08:
We currently support drag and drop, however the user needs to make sure he drags the image onto the first step. For better UX purposes, the user should be able to drag and drop to any part of the application.

lurch, 2016-11-08:
Merge this with #849 ?

jviotti, 2016-11-08:
I think they serve different clearly defined purposes, and thus make more sense as separate issues. This is about increasing the drag and drop area, while the other one about putting an overlay during hover. I guess https://github.com/resin-io/etcher/issues/849 somewhat depends on this one though.

jviotti, 2017-01-25:
By reading this again after a while, I agree with @lurch that it should be closed in favor of the other issue. Sorry for the non-sense :)

closed 2017-01-25 by jviotti