balena-io/etcher #850: Writing continues after GUI is closed

lurch, 2016-11-08:
- **Etcher version:**  1.0.0-beta16 (built from latest GH source on 2016-11-08)
- **Operating system and architecture:** Ubuntu Linux 14.04 x64
- **Do you see any meaningful error information on DevTools?** n/a

I've got an older USB drive which is quite slow (and which blinks an LED when it's being accessed - which is how I noticed this bug!).
I tried writing a big image to it, which started off okay, and then I closed the Etcher GUI because I couldn't be bothered to wait for it to finish. The GUI all closed down okay with no errors, but to my surprise the LED on the USB drive continued to blink! I checked with `top` and the `electron` process was still running in the background, continuing to write to the drive.

I launched Etcher with `npm start` and nothing got printed to the terminal apart from:
```

> etcher@1.0.0-beta.16 start /home/andrew/github/lurch/etcher
> electron lib/start.js

```

lurch, 2016-11-08:
Just tested the same behaviour on Windows10, and when closing the application window _all_ the Etcher processes die off, and the LED on the slow USB drive stops flashing (which is the expected behaviour).

jviotti, 2016-11-09:
So this seems to be an issue in https://github.com/jorangreef/sudo-prompt. I'll investigate a bit and report back my results soon.

lurch, 2016-11-09:
Were you able to replicate this on Mac?

jviotti, 2016-11-09:
Turns out yes, but its not very trivial. The process remains alive if we do SIGKILL, but not if we do SIGINT.

closed 2016-11-16 by jviotti