balena-io/etcher #851: Don't require admin password for multiple flashes

lurch, 2016-11-08:
- **Etcher version:** 1.0.0-beta16
- **Operating system and architecture:** Ubuntu Linux 14.04 x64

Slightly related to #847 ....

If I write an image to a drive, I have to provide my administrator password in order to elevate the child image-writer process. If I then choose to write the same image again, when I click `Flash!` I *again* get asked for my administrator password.
I wonder if the image-elevation process could be reworked slightly (sorry, I know it's already had **lots** of tweaks!) so that the admin password only has to be entered the first time you click `Flash!`, and not on each subsequent time? Maybe there could be a 'persistent' process that gets elevated the first time you `Flash!`, and which stays running in the background, and which kicks off the child-writer process the next time you click `Flash!`.

(If we're leaving an elevated process running in the background (and presumably communicating with it via IPC) I guess we'd need some kind of 'security' to prevent processes _other_ than Etcher from communicating with it?)

jviotti, 2016-11-09:
Yeah, this makes a lot of sense, and definitely something we want to implement. So as it is right now, Etcher spawns the child writer process, which then spawns itself for elevation, and then spawns the CLI. Instead of doing the whole process each time, we can keep the spawned child writer alive, and have some way of asking it to spawn CLI instances.

I agree we have to consider security very well here, but I don't have clear thoughts on that yet.

lurch, 2016-11-09:
Just been thinking a bit more about the security implications of this... perhaps we should only allow the IPC process to kick off an elevated child-writer (assuming there's not already another write in-progress of course) with a same or a different image, and not allow the IPC channel to change which drive we're writing to.
So if we flash the first image to `/dev/sdb`, and we want to flash the second image to `/dev/sdc`, then that **should** ask for the user's password again?  (which prevents a malicious process from trying to write a file to e.g. `/dev/sda` and destroying the user's HDD)

jviotti, 2016-11-09:
That makes a lot of sense. I'm happy to move forward with that approach!

jviotti, 2016-11-19:
Closing this, since the main use case for remembering elevation is users flashing the same image to many drives, which can be better solved with multi writes support.

closed 2016-11-19 by jviotti

lurch, 2016-11-28:
...or just launching Etcher with `sudo` ;-)