balena-io/etcher #852: Publishers' UI customisation tool

konmouz, 2016-11-08:


jviotti, 2016-11-09:
Some ideas:

- Create a CLI, or desktop application (even inside Etcher itself)
- Maybe expose a service with a nice wizard-like web UI that generates the final package and allows the user to download
- Create a tool that "lints" a manually built package (for more hardcore users)

konmouz, 2016-11-09:
Just to add:
- Simple UI customising tool with UI preview and CL style export of selected settings
- Implement instructions in a responsive and adaptive UI to accommodate relevant info and steps
- Accessible instructions without the need to flash an img

jviotti, 2017-01-25:
This is not going to be needed anymore given recent changes to the extended archive format.

closed 2017-01-25 by jviotti