balena-io/etcher #858: Display warning if image is too big when using CLI

lurch, 2016-11-09:
- **Etcher version:** 1.0.0-beta16
- **Operating system and architecture:** Linux x64

When using the Etcher CLI (see https://github.com/resin-io/etcher/blob/master/docs/CLI.md ) there's no check that the image is smaller than the drive size, which means it "allows" you to attempt to write e.g. a 3GB image onto a 2GB USB stick.

stefan-mihaila, 2016-12-08:
What's the desired behaviour? I see more options:

1. Allow the user to select a disk smaller than the image, and if he does, display a warning and prompt the user if he wishes to continue with the selected disk
2. Allow the user to select a disk smaller than the image, and if he does, bail out with an error without even trying to burn the image
3. Filter out (do not display) disks smaller than the image.

Option 3 is more work to implement, because from what I see, the resin-cli-visuals module is responsible for filtering out system drives: https://github.com/resin-io/resin-cli-visuals/blob/f3d643008be2cd6b5b0c51133080618a1d69d130/build/widgets/drive/index.js#L40. I don't think that the image size should get passed all the way to the visuals module.

lurch, 2016-12-08:
I guess it depends if we want the CLI to behave the same way as the GUI (which does option 3), or if we want the CLI to be a more 'hardcore' version, in which case I guess option 2 would be best?
I'm not sure why we'd ever want to allow option 1?

ping @konmouz @jviotti

jviotti, 2016-12-08:
I think it greatly depends on what we're doing with https://github.com/resin-io/etcher/issues/888, since both the CLI and the GUI should have a similar look 'n feel. 

At the time of this writing, the consensus on the issue linked before is that we will show every single drive on the drive selector modal (even if its non-removable, but making it very clear that that's the case), but we will present a warning message if the drive the user selected is not removable.

I honestly don't like how we're depending on `resin-cli-visuals` for this, and since the paradigm of drive selection is changing in the GUI, I feel we need to re-work this.

@stefan-mihaila How do you feel about re-working a CLI drive selection widget that mimics what the GUI will be in the future?

jviotti, 2016-12-08:
Yeah, I think we should try to mimic the GUI when possible. I propose showing all the drives, but not allowing selection of the ones that are small enough (and present a confirmation warning if selecting a system one).

lurch, 2016-12-08:
...which I guess means we'd also need some way for the CLI to display 'flags' for protected / system drives? (currently there's no way to even force the CLI to display system drives)

lurch, 2016-12-08:
...and drive-multi-selection in the CLI could be an interesting problem too ;-)

jviotti, 2016-12-08:
@lurch Yeah, exactly. The whole widget needs to be re-designed, since its falling behind a lot, and we need to make this ready for the release.

@stefan-mihaila BTW, we'll be properly releasing the Etcher CLI soon (for v1.0), so polishing the whole CLI experience is something that needs a lot of love if that's something that interests you!

stefan-mihaila, 2016-12-08:
@jviotti Yes, it interests me 💯 😃

stefan-mihaila, 2016-12-08:
I'll start by showing all drives, display a warning if it's a system drive, and not let the user select smaller than image disks. @lurch has a point  with multi-selection (I assume it will be required for parallel writing), we'll see how that plays out...

konmouz, 2016-12-08:
Ideally the CLI and GUI should behave in a similar fashion. I agree that all detected drives should be displayed and then provide a warning (while selecting not flashing) if need be. Also, Etcher shouldn't auto-select a drive that is smaller than the selected image.

lurch, 2016-12-08:
> Also, Etcher shouldn't auto-select a drive that is smaller than the selected image.

I've just checked, and it doesn't. Phew!

jviotti, 2017-03-24:
I'm sending a quick PR to solve the main issue reported by this ticket: flashing an image that is larger than the drive. We will also keep working on the PR that provides a smarter drive detection CLI widget, with flags, etc.

closed 2017-03-28 by jviotti