balena-io/etcher #859: Error on apt-key

Cascador, 2016-11-10:
Hello,

sudo apt-key adv --keyserver pgp.mit.edu --recv-keys 379CE192D401AB61 doesn't work.

The key is missing on pgp.mit.edu ?

Thanks.

lurch, 2016-11-10:
Works fine for me? What error message do you get?

Cascador, 2016-11-10:
Problem solved with : sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61

Probably a firewall or proxy problem, see here : http://unix.stackexchange.com/questions/75892/keyserver-timed-out-when-trying-to-add-a-gpg-public-key

Thanks.

lurch, 2016-11-10:
Thanks for letting us know @Cascador .

@jviotti @dlech should this info be added to https://github.com/resin-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64 ?

Cascador, 2016-11-10:
sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61 works on more cases (especially at work). 

FYI in Docker sudo apt-key adv --keyserver hkp://p80.pool.sks-keyservers.net:80 (https://docs.docker.com/engine/installation/linux/ubuntulinux/)

PR #861 (2016-11-13): docs: more explicit key server
Use full URL with port for key server. Also use port 80 to work around firewalls and proxies.

Fixes: #859
commit f80feb9e (2016-11-13): docs: more explicit key server (#861)

Use full URL with port for key server. Also use port 80 to work around firewalls and proxies.

Fixes: #859
closed 2016-11-13 by jviotti