balena-io/etcher #862: False virus detected?

jridder, 2016-11-10:
- **Etcher version:**
- **Operating system and architecture:**
- **Do you see any meaningful error information on DevTools?**

Etcher 1.0.0-beta.16
Windows 10 Pro build 14393.447

I have Webroot installed and when I click on the flash button Webroot complains that a file with Trojan.Dropper.Gen was written into the temp folder. Can someone confirm that there that the files created are clean before I go to Webroot?

jviotti, 2016-11-11:
We had similar problems with other virus scanner programs in the past, and the solution is to whitelist our code signing certificate there. Since the virus scanner doesn't know about us, it fallbacks to throwing some warnings.

I wrote an email to Webroot support, so we'll probably hear from them soon.

closed 2016-11-11 by jridder

jviotti, 2016-11-12:
Hi @jridder ,

The Webroot team responded, and they've said that they download the application but didn't get any detections. They told me that if you can send them the detailed report, they could assist further.

Thanks a lot!