balena-io/etcher #870: Windows 10 x64: Error: 0x80131700 when listing drives

jviotti, 2016-11-16:
- **Etcher version:** https://github.com/resin-io/etcher/pull/869/commits/41c413525bd691d91433d3ae58013ccef5b06454
- **Operating system and architecture:** Windows 10 x64

I get this error right when opening Etcher from the source tree, with `npm start`:

[image]

The drive scanner mechanism doesn't work at all. This is a clean Windows 10 installation on an Intel NUC.

jviotti, 2016-11-16:
The `win32.bat` line pointed out by the error corresponds to:

```vbs
Set GetTopLevelDrives = CreateObject("System.Collections.ArrayList")
```

See https://github.com/resin-io-modules/drivelist/blob/master/scripts/win32.bat

jviotti, 2016-11-16:
See https://blogs.msdn.microsoft.com/dsvc/2013/03/04/usage-of-net-collections-types-in-vbscript-is-not-supported-after-net-4-5/

> Now let me explain the reason of this failure. When “CreateObject()” is called, it internally invokes the COM call CoCreateInstance() to create an ArrayList object. The way CoCreateInstance() works is it first tries to look into the system registry to find the assembly which implements this ArrayList type identified by its ProgID i.e. “System.Collections.ArrayList” (in this case, the ProgID and the type name are same for ArrayList). As .NET 4.5 which is the only .NET Framework inbuilt with Win2K12 has stopped exposing .NET types as COM interfaces, the registry lookup by CoCreateInstance() just fails. However, Windows seems aware of this change, and thus tries to install .NET 3.5 which does expose .NET types as COM interfaces and shows the above popup message. Once .NET 3.5 is explicitly installed on these machines, the script starts working just fine as it works on the older operating systems.

> Considering the complete Framework overhaul from .NET 2.0/3.0/3.5 (i.e. CLR v2) to .NET 4.0/4.5 (CLR v4), I think, it’s completely a design decision to stop exposing these types as COM interfaces to prevent breaking the older VBScript/native apps already in the market which are being targeted to .NET 3.5 and use these kind of .NET types. Had the .NET 4.5 exposed these types as COM interfaces, and in case the .NET 3.5 also gets installed on the machine side-by-side, the same COM interfaces (something like, “System.Collections.ArrayList”) exposed by two versions of the Framework would be conflicting to each other. And the fact that multiple versions of the .NET frameworks could/should get installed side-by-side on a machine, COM registration of the .NET types would be a bad idea, hence it has been stopped.

> Having said that, there are some options as followed that might be helpful in these kinds of situations –

> – Install .NET 3.5 (which is the recommended & the best option ever).

> - Don’t use .NET types and rather implement your own types in VBScript/native languages (in this case, ArrayList).

> – Create a managed COM object wrapper around the .NET type of interest and consume it from VBScript/native languages.

> – Force cscript.exe to use .NET 4.5 by creating an activation config file as below (i.e., v4.0 for the inbuilt .NET 4.0/4.5 on Win2K12):

```
        <configuration>
            <startup useLegacyV2RuntimeActivationPolicy="true">
                <supportedRuntime version='v4.0' />
             </startup>
         </configuration>
```

lurch, 2016-11-16:
IMHO given the already-large size of the Etcher download, we shouldn't require people to install .NET 3.5 too.
How hard is the "implement your own types in VBScript" option?

lurch, 2016-11-16:
Just found http://stackoverflow.com/questions/13585660/lists-in-vbscript - dunno if it's any good ;)

jviotti, 2016-11-16:
> IMHO given the already-large size of the Etcher download, we shouldn't require people to install .NET 3.5 too.

I definitely agree. The `List` class proposed in one of the SO answer looks worth trying.

jviotti, 2016-11-16:
The proposed `List` class from the answers can't handle object items (like dictionaries). I ended up rolling my own based on `Scripting.Dictionary`, as suggested in yet another answer, and it works flawlessly :+1:

closed 2016-11-16 by jviotti