balena-io/etcher #874: Pressing escape key during drive-selection dialog displays error window

lurch, 2016-11-17:
- **Etcher version:** 1.0.0-beta16 (latest from github)
- **Operating system and architecture:** Linux x64
- **Do you see any meaningful error information on DevTools?** 
```
tracker.js:10  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io
tracker.js:10 Thu Nov 17 2016 12:11:57 GMT+0000 (GMT) Application start
tracker.js:10 Thu Nov 17 2016 12:11:57 GMT+0000 (GMT) Checking for updates
tracker.js:10 Thu Nov 17 2016 12:12:21 GMT+0000 (GMT) Select image ({"size":1389363200,"path":"/home/andrew/Downloads/2016-09-23-raspbian-jessie-lite.zip","logo":false,"bmap":false})
tracker.js:10 escape key pressa.(anonymous function) @ tracker.js:10
```

Steps to reproduce:
 1. Connect 2 USB drives to your computer
 2. Start Etcher
 3. Select an image as normal
 4. Click on the "Select drive" button
 5. Instead of selecting a drive, press the Esc button on your keyboard
 6. An error window pops up saying `"escape key press"`

I'd expect the Escape key to simply behave the same as clicking on the X to close the modal dialog.
(I checked, and the Escape key *does* work to close the "unsafe mode" dialog, with no errors being displayed)

lurch, 2016-11-17:
By running a quick `grep` over my local build directory, it seems the `'escape key press'` string is coming from `node_modules/angular-ui-bootstrap`.

lurch, 2016-11-18:
Same bug present on Windows too.

jviotti, 2016-11-18:
I remember @Shou encountered and fixed this issue as part of another PR. @Shou how do you feel about extracting this particular change as a separate PR that closes this issue, and then rebase such PR?

jviotti, 2016-11-18:
Indeed, see https://github.com/resin-io/etcher/pull/868/files#diff-3336b717ca478ff1594f6c3316f65a63R65

closed 2016-11-18 by jviotti